osrf/subt #647: Data missing from downloadable logs

zwn, 2020-10-05:
There is a discrepancy in what we see in WebUI and what we get to download. While the simulation is running we can press "Get logs" in the download menu and some kind of screen appears. When the simulation crashes, the stack trace can be seen in this screen. However, when the simulation is finally marked as finished, the contents of this screen are not part of the logs available for download. Actually, the only indication that the simulation most likely crashed is the fact that the logfile `subt_cave_<utc_time>.log` 
does not contain ending similar to this:

```
1509 484000000 finished_elapsed_real_time 4417 s.
1509 484000000 finished_elapsed_sim_time 1423 s.
1509 484000000 finished_score 3
```

nkoenig, 2020-10-07:
The live logs come from the Kubernetes pod itself. If a segfault occurs, then the segfault information is lost.

We may be able to mount the appropriate directory into the log copy pod, and save segfault information. However, this won't be done for Cave Circuit.

nkoenig, 2020-10-19:
I'll mark this an enhancement to increase the set of logs available for download.