osrf/subt #648: Logs missing on recent cloudsim run

realdealneil, 2020-10-05:
Team collembola (SSCI) recently posted a submission to cloudsim and it appears to have run without issues, but we are unable to download the logs.  Can you look into whether there's a way to access the logs?  The ID of the run was 651fbfad-1ffc-468b-84ee-fe14f4569857 and our title for it was ugv856uav939v8s2r1.  Thanks!

wolfgangschwab, 2020-10-06:
I had the same problem that I could not download the logs after I used the great functionality to View a simulation. When I use the "View simulation" on one run and stop it by using the arrow to the left on the upper left corner, I also have to close the webpage in the browser and reopen it in a new page. Otherwise I would have problems to start the download of the logs or even start another View simulation. It might depend on the used web browser. I'm using Firefox. But you can give this workaround a try.

angelacmaio, 2020-10-06:
@realdealneil your logs should be accessible now.
@wolfgangschwab thanks for posting #649 , will address as a separate issue.

closed 2020-10-06 by angelacmaio

realdealneil, 2020-10-06:
Thanks!  I can confirm that I'm able to download the logs now.  Much appreciated!