osrf/subt #650: No default constructor for subt::CommsClient class

pauljurczak, 2020-10-06:
ROS subscribers and publisher classes provide a default constructor, so they can be properly initialized inside the node constructor body as shown in many examples of ROS node implemented as a C++ class. For example:
```
  public:
    Publisher() {}
```
in ROS `publisher.h` source.

Can default constructor be added to `subt::CommsClient` class, so I can initialize all data members uniformly inside the constructor body, instead of a bit inflexible constructor member initializer list:
```
Node::Node(string robotName) : commsClient(subt::CommsClient(robotName)) {........}
```

peci1, 2020-10-08:
Store the client in a unique_ptr.

pauljurczak, 2020-10-08:
@peci1 That's what I'm doing now, but I would prefer better uniformity and it may save other people some time in the future. Not a big problem in the grand scheme of things, though.