osrf/subt #652: Allow more runs to be queued on cloudsim (limit running not queuing)

zwn, 2020-10-07:
Currently we can only submit as many simulation as is the limit per team on running simulations. It would be advantageous being able to queue up more simulations to be executed in sequence. So we are proposing to change implementation of the per team limit from limiting submissions to limiting the parallel runs only.

Even better would be to limit the number of AWS machines alocated per team instead.

https://github.com/osrf/subt/issues/644#issuecomment-704512723

AravindaDP, 2020-10-18:
I would disagree with this proposal and believe current approach is more fair. (Given that I believe simulations are queued in a global queue)

If allowing unlimited queuing is desired, the implementation should be such that each team should have own queue and simulation runner should pick a pending simulation from each team in a round robbin fashion.

zwn, 2020-10-19:
@AravindaDP I believe you might have misunderstood the nature of the proposal. I am not proposing a way for a team to gain advantage in access to the shared cloudsim. I am saying that babysitting the runs is not an effective use of our time.

That said, I was working with the assumption that there are enough AWS resources that each team can have 3 simulations running at any time (which seemed to be the case). If this assumption is correct, than the queuing mechanism does not matter wrt fairness.

AravindaDP, 2020-10-19:
@zwn Apologies I think I misinterpreted your proposal.

It seems both of us are actually referring to same end result with different ways of implementing.