osrf/subt #653: Camera rendering issues in cave practice 01

malcolmst, 2020-10-07:
I haven't been looking at much of the raw camera data lately with the cave practice worlds, but just took a close look at some bag data gathered from `980f509e-4ab2-4170-9675-485ae7ea2d27` (cave practice 01). I'm seeing some really weird rendering and lighting/shadow issues, especially near the entrance, but also going into the cave as well. This is just taken from the compressed camera topics.

The following is camera data from an X2 robot entering the cave:

[image]

And another X2 robot waiting outside the entrance, then entering:

[image]

Then some strange lighting effects inside the cave near the first rope artifact:

[image]

I'll keep a lookout for any similar effects in the other worlds.

malcolmst, 2020-10-07:
The super-brightness seems to come and go really quickly depending on the angle to a nearby obstacle:

[image]

nkoenig, 2020-10-07:
I believe this is a duplicate of issue #635

malcolmst, 2020-10-07:
Yes I think you’re right, sorry I missed that

nkoenig, 2020-10-08:
Duplicate of #635.

closed 2020-10-08 by nkoenig