osrf/subt #656: Playback of Old Cloudsim Logs

DalInar, 2020-10-07:
Hello!  I'm running into an issue with playing back cloudsim logs from back during the urban circuit through ignition.  I'm following the instructions here: https://github.com/osrf/subt/wiki/Cloudsim%20Log%20Playback .  I am trying to playback logs locally, not through a docker image.

The playback works fine when I use a state.tlog file from a more recent simulation (I've succeeded with outputs from recent cave circuit tests), but it fails when I try to use a log file from the urban competition.  

For example,
 `LC_ALL=C ign gazebo -v 4 -r --playback ~/subt_logs/subt_urban_logs/01/logs/`

Opens Gazebo, but with a blank world:
[image]

In the terminal, I suspect the issue has to do with the following out:

[Msg] Found no publishers on /clock, adding root clock topic
[Dbg] [SimulationRunner.cc:402] Creating PostUpdate worker threads: 2
[Dbg] [SimulationRunner.cc:415] Creating postupdate worker thread (0)
Database is corrupt, retrieving last valid message.Playback may fail or be truncated.

Is there a way to modify the old logs to get around the database corrupted error?  Or do I need to revert back to old versions of ignition?  Thanks!

JShep1, 2020-10-20:
Hi @DalInar , what system (Bionic, Focal, etc.) and version of `ign gazebo` are you running, you can run `ign gazebo --versions` to get the version number.  Additionally, can you verify that this `state.tlog` file is valid and able to be run (possibly by either attempting to run with a previous version of `ign gazebo`, or trying another logging example on your current version of `ign gazebo`).  I'm mainly trying to determine if this is an issue with the file itself or a possible bug within `ign gazebo`.

nkoenig, 2021-01-13:
Closing due to inactivity. All refer to issue #328

closed 2021-01-13 by nkoenig