osrf/subt #659: Ground Truth TF's slow after update

dan-riley, 2020-10-10:
Since updating to the latest Ignition and SubT versions (as detailed in the update page), enableGroundTruth is essentially broken.  We have tried several different robot models, and they all publish extremely slow TF's (around 0.5-1hz it seems) and there seems to be some other massive cpu or memory usage happening. When running with ground truth off everything runs as it did before.  I'm not sure which update caused the change, but we're seeing it on multiple machines.

nkoenig, 2020-10-19:
@dan-riley , did @acschang answer solve your problem?

dan-riley, 2020-10-19:
@nkoenig Yes, this seems to work.  We used 50hz.  I agree that it should probably be moved off of the static TF.

peci1, 2021-01-19:
Solution proposed in #759.