osrf/subt #660: Cloudsim run halted early on indicating no errors

realdealneil, 2020-10-10:
We just posted a run that only lasted about 100 seconds in sim but it terminated and indicated that there were no errors.  We are confused about what's happening...it is from Team Collembola's run with the following ID:

e108c26c-09a7-4acd-b4af-799356a7f762

Our name for this run was svn969v8p1r1.  

The run.yml file indicates the following:

sim_time: 101
real_time: 1112

sim_time should be 3600 seconds.  Yet, the status in the Compete tab indicates:

Terminated
With no errors

We would like to know what's going on in case we're doing something wrong, though we've used this identical set of docker images on other circuits without problems like this, so we don't think it's on our end.

rsawtell, 2020-10-10:
I've had this happen to several of our simulations: https://github.com/osrf/subt/issues/639, so it's not limited to your solution. Like yours, the same set of images have worked fine in other runs.

realdealneil, 2020-10-10:
I would be less worried if the run had indicated that there was an error during the simulation.  I just don't want runs like this to count against us in the actual competition.

peci1, 2020-10-10:
We had these 100-600 second runs when we sent a bad docker image. Apparently, if there are no running robot containers except TEAMBASE, the simulation is halted. There's no way how you could get a log of the failed `docker run` command. But as it is docker, it should be easy to call the `docker run` yourself on your computer on the same images. Usually, if there's an error starting the container, you'd detect it this way.

If you're sure the container runs correctly on your machine, then it could indicate a bug in cloudsim.

realdealneil, 2020-10-11:
It's definitely not a problem with bad docker images.  They have run correctly on all of the circuits multiple times.

rsawtell, 2020-10-12:
Same for us, we've used the same images successfully, often at the same time as the run that failed was running. Additionally we saw in the live viewport the robots had already begun exploring at the time the problem occurred. I would agree that the primary concern should be that these sort of runs should not count in the competition, and they are easily identifiable by the fact the simulation time is not 3600.

peci1, 2020-10-12:
I think it would really help debugging these issues if there were a field indicating the reason why the mission ended. The options are e.g. calling /subt/finish, running out of time, exhausting artifact reports, finding all artifacts ( :) ), ale dockers exiting (with zero or non-zero exit status) etc... It'd be great to see these reasons.