osrf/subt #661: Bridge crashed in subt_ros_relay ASSERTION FAILED Call to publish() on an invalid Publisher

zwn, 2020-10-10:
While running locally the bridge container, it crashed with the following output on the screen:

<pre><font color="#34E2E2"><b>[Dbg] [Manager.cc:443] Death of process[59] with name[x4_description].</b></font>
<b>process[X150L/ros_ign_bridge_imu-1]: started with pid [105]</b>
<b>process[ros_ign_bridge_larsa_58_1154408817406722280-2]: started with pid [112]</b>
<b>process[X150L/ros_ign_bridge_gas-2]: started with pid [134]</b>
<b>process[subt_ros_relay-3]: started with pid [147]</b>
<font color="#CC0000">[FATAL] [1602334699.294313202]: ASSERTION FAILED</font>
	<font color="#CC0000">file = /opt/ros/melodic/include/ros/publisher.h</font>
	<font color="#CC0000">line = 103</font>
	<font color="#CC0000">cond = false</font>
	<font color="#CC0000">message = </font>
<font color="#CC0000">[FATAL] [1602334699.300362028]: Call to publish() on an invalid Publisher</font>
<font color="#CC0000">[FATAL] [1602334699.300386828]: </font>

<font color="#CC0000">[subt_ros_relay-3] process has died [pid 147, exit code -5, cmd /home/developer/subt_ws/install/lib/subt_ros/subt_ros_relay __name:=subt_ros_relay __log:=/home/developer/.ros/log/43a7f58e-0af8-11eb-ae68-48a472fb7d13/subt_ros_relay-3.log].</font>
<font color="#CC0000">log file: /home/developer/.ros/log/43a7f58e-0af8-11eb-ae68-48a472fb7d13/subt_ros_relay-3*.log</font>
</pre>

It seems it could be some kind of race condition.

nkoenig, 2020-10-13:
What command did you use to produce this?

zwn, 2020-10-14:
I used 3 terminals. In the first one I started the simulation:
```
./subt/docker/run.bash osrf/subt-virtual-testbed:cloudsim_sim_latest \
  cloudsim_sim.ign \
  headless:=$HEADLESS \
  seed:=1 \
  circuit:=$CIRCUIT \
  worldName:=$WORLD \
  robotName1:=$ROBOT \
  robotConfig1:=$CONFIG
```
In the second one the bridge:
```
./subt/docker/run.bash osrf/subt-virtual-testbed:cloudsim_bridge_latest \
  circuit:=$CIRCUIT \
  worldName:=$WORLD \
  robotName1:=$ROBOT \
  robotConfig1:=$CONFIG
```
In the third one I started our solution. Something didn't look right (not seeing the right messages on the console) so I switched to the other two (I almost never look at the bridge console since it never outputs anything after starting) and there was the crash.

The parameters were set to headless=1, circuit=cave, world most likely cave simple 01 (but I didn't take a note) and robot was SSCI_X4_SENSOR_CONFIG_2.

We are reading the robot description as a first thing while starting our solution while doing some detective work finding what type of robot we are supposed to be driving.

The docker images are the latest release 2020-10-07.