osrf/subt #662: Marble HD2 pan_tilt camera partially clipped top 180 lines of image

knoedler, 2020-10-11:
[image]
The top 180 lines or so of the Marble HD2 pan tilt camera are clipped for both the image and depth views.  It is not completely clipped as weird echo images sometimes show up in the top part.  Further it is not clipped exactly on a line as you can see the first unclipped line is partial.

To repeat: (the issue seems to appear in every world)
ign launch -v 4 cave_circuit.ign worldName:=cave_qual robotName1:=MHSC2 robotConfig1:=MARBLE_HD2_SENSOR_CONFIG_2
view the pan_tilt image in rviz

knoedler, 2020-10-11:
This issue is not blocking me, so my preference would be to leave it alone it until after the cave circuit.

arjo129, 2020-10-13:
I can confirm the clipping behavior, but it clips by a different amount for me (around 20pixels). There's also clipping on the bottom RGBD camera.

knoedler, 2021-03-09:
Bumping this issue as the clipping has gotten worse and is sometimes over half of the image.

nkoenig, 2021-03-18:
The cause is due to bad placement of the camera.

Current camera placement highlight by red sphere:
[image]

New camera placement highlight by red sphere:
[image]

See pull request #827

closed 2021-03-22 by nkoenig