osrf/subt #663: Segfaults in `cave_practice_03` world

arjo129, 2020-10-13:
```
[Dbg] [VisibilityRfModel.cc:80] Range: 15.4046, Exp: 1.5, TX: 20, RX: -37.8148
[Dbg] [VisibilityRfModel.cc:80] Range: 0, Exp: 1.5, TX: 20, RX: inf
[Dbg] [VisibilityRfModel.cc:80] Range: 15.4911, Exp: 1.5, TX: 20, RX: -37.8512
[Dbg] [VisibilityRfModel.cc:80] Range: 0, Exp: 1.5, TX: 20, RX: inf
[Msg] Unloaded level [82]
./run_sim.bash: line 6:    51 Segmentation fault      (core dumped) ign launch -v 4 $@
```

How to reproduce. Start at staging. Turn right before entering the cave and head towards the grass patch on the side using marble husky.

adlarkin, 2020-10-14:
I was unable to reproduce this issue. I tried using the [osrf/subt-virtual-testbed](https://hub.docker.com/r/osrf/subt-virtual-testbed) Docker image and the [subt_shell](https://github.com/osrf/subt/blob/master/docker/subt_shell/Dockerfile) Docker image.

@arjo129 is this segfault something you are seeing consistently, or have you only seen it once? Perhaps it was an edge case/race condition, or something related to your machine's computing power/resources. Would you mind trying both the `osrf/subt-virtual-testbed` image and `subt_shell` image to see if this issue appears in either of these settings? When running the `osrf/subt-virtual-testbed` image, please make sure to run `docker pull osrf/subt-virtual-testbed:latest` first in order to get the most up-to-date Docker image.

arjo129, 2020-10-15:
Its a problem with `subt-virtual-testbed:latest` (I pulled it last on wednesday so it should be up to date). I can investigate a bit more after my submission!

adlarkin, 2020-10-15:
> Its a problem with subt-virtual-testbed:latest (I pulled it last on wednesday so it should be up to date)

Looking at the [history for the `latest` tag](https://hub.docker.com/layers/osrf/subt-virtual-testbed/latest/images/sha256-7a43349a61330384c96f837f7b562a61de5d6de118f703e3a93ed3cce094363e?context=explore), it looks like the last push was 6 days ago, which I believe means last Thursday - so, there's a chance that the image you have may not be the newest one. But yes, please let me know how things look after another pull and investigation - if it's still an issue, I can continue to work with you on how to get this resolved.

adlarkin, 2020-10-29:
@arjo129 any updates on this? Have you been able to re-produce this issue using the latest `subt-virtual-testbed` image? If not, I'd like to close this issue.

arjo129, 2020-10-31:
No sorry for the delayed response. Yes it seems fixed in newer images.

closed 2020-10-31 by arjo129