osrf/subt #664: RSSI reported by robot seems to be infinity

arjo129, 2020-10-13:
```
[Dbg] [VisibilityRfModel.cc:80] Range: 12.3693, Exp: 1.5, TX: 20, RX: -36.3852
[Dbg] [VisibilityRfModel.cc:80] Range: 2.00437, Exp: 1.5, TX: 20, RX: -24.5297
[Dbg] [VisibilityRfModel.cc:80] Range: 0, Exp: 1.5, TX: 20, RX: inf
[Dbg] [VisibilityRfModel.cc:80] Range: 12.3693, Exp: 1.5, TX: 20, RX: -36.3852
[Dbg] [VisibilityRfModel.cc:80] Range: 2.00437, Exp: 1.5, TX: 20, RX: -24.5297
[Dbg] [VisibilityRfModel.cc:80] Range: 0, Exp: 1.5, TX: 20, RX: inf
```
So I have a teambase and one husky. If I call `get_neighbours` on teambase it gives me the rssi to the robot. On the other hand the robot reports an RSSI range of Infinity. Not sure what to make of this.

arjo129, 2020-10-13:
Duplicate of #643

closed 2020-10-13 by arjo129

peci1, 2020-10-13:
And do you actually read the robot the `inf` RSSI is associated with? Because when you publish to broadcast, you also publish to yourself (with `inf` RSSI).