osrf/subt #667: bottom_scan not being published

zwn, 2020-10-14:
In the run `7090d419-5ee0-419c-abfd-ae8f845b81aa` the topic `/A100W900LH700R3600W/bottom_scan` has never been published (SSCI_X4_SENSOR_CONFIG_2).

There is no mention of this topic in `rostopic_stats_logger.log` while at least on one other robot in this run (B700W600LH400R3600W) it is a 5th entry in the file.

The process responsible for the topic seems to have been started:
```
260:[roslaunch][INFO] 2020-10-13 14:34:08,476: process[A100W900LH700R3600W/ros_ign_bridge_gpu_lidar_bottom-15]: started with pid [348]
```
and atop log shows it is running. But it is completely missing from `rostopic_stats_logger.log`.

Needless to say - without this data we are screwed.

m3d, 2020-10-15:
In the simulation from the last night the drone A did not start (the propellers did not turn in visualization, while B drone left staging area OK) although according to our logs `cmd_vel` was send (robotika, Cave Practice 2, `eaee7507-5006-4529-a2c0-1f283c9481b0`). As @zwn said, if this connection is broken that would be fatal :(. Are there any instructions what we can in such a case check? Please let us know (bridge logs, sim logs?)

m3d, 2020-11-01:
Maybe I should mention the problem with bottom lidar here too: https://github.com/osrf/subt/issues/670#issuecomment-720053628 (Urban 3, robotika, ver92u3, 1144a21f-517f-439c-91a3-9d15ce8be2e9)