osrf/subt #668: Question about submission process

malcolmst, 2020-10-15:
I haven't tried to submit yet and I'll probably be working on this up to the last minute, but wanted to check, does the submission count towards the limit of 3 simultaneous running simulations, or will the submitted simulation go away to allow more to be run?

m3d, 2020-10-15:
According to our experience (we tried the submission process and re-submission) last night, you can still run 3 simulations in parallel. The submission is in `Pending With no errors` and the replaced submission is in `Superseded With no errors` state.

malcolmst, 2020-10-15:
Thanks @m3d!

m3d, 2020-10-15:
Well, a couple of hours later **Launching With no errors** that is surely not correct and it blocks other simulations now!

m3d, 2020-10-15:
That was a strange "glitch" - unfortunately I did not make screenshot ... and now both submissions are in pending state.

angelacmaio, 2020-10-15:
Cave Circuit competition submissions do not count towards the limit of 3 simultaneous running practice simulations. Final submissions display the status "Pending" in the SubT Virtual Portal, whereas all previous Cave Circuit submissions are marked as "Superseded".

closed 2020-10-15 by angelacmaio

malcolmst, 2020-10-15:
Thanks @angelacmaio for clarifying, and thank you for the new "stop" button! That will be incredibly helpful to avoid having to ration simulations :).