osrf/subt #669: drone rotors never start turing

zwn, 2020-10-15:
In simulation `eaee7507-5006-4529-a2c0-1f283c9481b0` the drone named A600L never took off.

When analyzing the problem, I was able to confirm that some speed commands got all the way to the process `/A600L/ros_ign_bridge_twist` - see this excerpt from `rostopic_stats_logger.log`:

```yml
topic: "/A600L/cmd_vel"
node_pub: "/A600L_proxy"
node_sub: "/A600L/ros_ign_bridge_twist"
window_start: 
  secs: 86
  nsecs:         0
window_stop: 
  secs: 89
  nsecs:         0
delivered_msgs: 3
dropped_msgs: 0
```

Further analysis of the file `state.tlog` revealed that the drone rotors never started to turn. In the file in the "pose" message there are two poses with the name `rotor_0` and different ids - 175 and 224. The rotor with id 224 starts to move at around 92s which directly corresponds to our commands to the other drone in the simulation. The rotor with id 175 never turns (its orientation never significantly changes except some noise).

nkoenig, 2020-10-16:
At 1:32 sim time, B10W600R spins up, and A600L never starts. I can't see any problems in the logs. I assume you sent the same command to both robots?

zwn, 2020-10-19:
>  I assume you sent the same command to both robots?

The `rostopic_stats_logger.log` notes (quoted above) that the command from our node has been sent to the bridge an 86s of sim time. AFAIU any twist command (no matter the value) starts the rotors. But yes, we send the same commands - the only difference should be 10s wait for the B robot wrt the A robot.