osrf/subt #670: Cloudsim still drops messages

zwn, 2020-10-15:
It seems we still have the problems mentioned in #246 or #261. Taking into account our recent problems (for details see #655, #667, #669) I think it is safe to say that the problem has not been resolved yet. It (again) seem to be related to the intensity of cloudsim usage - the problems started to appear relatively recently and the frequency seems to be increasing (we might not be catching everything and the analysis takes valuable time).

rsawtell, 2020-10-15:
We concur, we're seeing sensor data such as LIDAR drop out at random and performance has been very hit or miss. At least for the last 2 days there has been a stark contrast between performance at night and during the day (US hours)

nkoenig, 2020-10-15:
@zwn  and @rsawtell , can you point me to specific instances where you saw these errors?

zwn, 2020-10-15:
@nkoenig I have posted several issues with very exact descriptions of what has happened. They are linked from the main text in this issue - #655, #677, #669.

arjo129, 2020-10-16:
> We concur, we're seeing sensor data such as LIDAR drop out at random and performance has been very hit or miss. At least for the last 2 days there has been a stark contrast between performance at night and during the day (US hours)

Yes just had an issue with our SLAM drifting like crazy cause no LiDAR data arrived >.<.

zwn, 2020-10-16:
It seems that the likelihood of this problem increases with the number of robots in the simulation.

m3d, 2020-11-01:
I can confirm that on CloudSim the problem still persists. Test in Urban 3 World, one drone (A600L) was not receiving any `bottom_scan` data (ver92u3, `1144a21f-517f-439c-91a3-9d15ce8be2e9`).

zwn, 2020-11-02:
The last run with this problem is only with 2 robots and the load on cloudsim is likely low right now, so that might rule out the previous guesses on the likelihood of reproduction.

m3d, 2020-11-02:
> The last run with this problem is only with 2 robots and the load on cloudsim is likely low right now, so that might rule out the previous guesses on the likelihood of reproduction.

Yes, it was only 2 robots, but I suppose that at the moment is running Cave Circuit evaluation, i.e. potentially large load?

zwn, 2020-11-02:
> I suppose that at the moment is running Cave Circuit evaluation, i.e. potentially large load?

I am hoping the evaluation runs on a separate installation of the cloudsim environment, that is not accessible to anyone from outside, thus without the risk of interference. I consider the instance accessible to competitors a proving ground for the real thing.

@nkoenig @angelacmaio  Is that correct expectation?

zwn, 2020-11-03:
btw: can this https://github.com/ignitionrobotics/ign-transport/issues/192 be related?

angelacmaio, 2020-11-04:
@zwn The Cloudsim instance that initiates competition runs is isolated from the practice Cloudsim instance.

peci1, 2021-01-18:
I also hope ignitionrobotics/ign-transport#192 would help. When do you plan updating the dockers in cloudsim to include it?

peci1, 2021-08-24:
Caught in action. In 2989aaca-1bb3-455d-a61f-469e4463935c , there was not a single message on `/X3/points` topic. File bridge_logger--X3-points.log proves that.

@nkoenig I think it would be fair to give teams the opportunity to specify a list of bridge topics they rely on, and if there would a topic missing from this list in a simulation, the simulation should be scratched and re-run for the finals evaluation.

peci1, 2021-08-24:
In server_console, I see 

    [Msg] Publishing laser scans on [world/finals_practice_02/model/X3/link/encoder_rotating_link/sensor/gpu_lidar/scan]

So the Ignition topic has been definitely at least advertised. Thus I'd blame the Ign-ROS relay.

I also connected to this simulation via web visualization and tried to display the missing points topic. Even cloudsim did not display it. Together with the previous point, it seems the Ign publisher had some problem either registering subscribers or publishing the messages to them.