osrf/subt #672: Cloudsim re-adding runs and stop button not working

realdealneil, 2020-10-15:
We have noticed a few times when cloudsim has run some of our runs more than once even though we only uploaded once. (Team Collembola).  One example is tbs1014qualr1 (ID: 356639c0-6344-4e57-8dbb-c87981b0054a-r-1).  Note that it ran yesterday and then reran itself today (has the exact same ID both times).  This wouldn't be a problem if we could stop the runs (cancel them), but the cancel button doesn't seem to do anything.  We get a confirmation screen, we hit ok, but the run keeps on going.

malcolmst, 2020-10-16:
The cancel button is working for me now, though I was also having some trouble with it not working earlier. I think the r1 runs usually happen automatically when the first run fails (sometimes anyway).

nkoenig, 2020-10-16:
I'll close this issue because the cancel button is working.

closed 2020-10-16 by nkoenig