osrf/subt #673: Comms not present in a simulation

malcolmst, 2020-10-15:
I just had an issue when running a cave simple 1 scenario (running on a local system) that comms was not present between any of the robots, and they were not able to report artifacts. I didn't see any obvious errors in the logs, but I did notice one thing in `events.yml`: it seems one of the robots dropped a breadcrumb before leaving the staging area. 

```
- event:
  id: 5
  type: breadcrumb_deploy
  time_sec: 117
  robot: X2N2
- event:
  id: 6
  type: detect
  time_sec: 120
  detector: staging_area
  robot: X2N2
  state: exit
```

Could this be the cause? Happy to provide any additional logs.

malcolmst, 2020-10-16:
Also, if this is just some sporadic problem, is this something that could be monitored during the evaluation runs?

acschang, 2020-10-16:
I just attempted to reproduce this locally in `cave_simple_01` and I was unable to do so.

The 3 cases after I dropped a vehicle in the staging area:
- Communication between both robots (A and B) in the staging area
- Communication between robot A in the first tile and robot B in the staging area
- Communication between robot B in the first tile and robot A in the staging area
- Communication between both robots (A and B) in the first tile

malcolmst, 2020-10-16:
Thanks for checking @acschang! I am seeing what appears to be a similar problem in the currently running simple 1 simulation `e5fcd870-64d5-43a1-8d7f-99ee372c838e`, though it was able to successfully report an artifact in this case. I'll take a closer look at those logs shortly. Doesn't appear to be anything unusual about the breadcrumbs in that scenario, but the robot behavior indicates they are not communicating.

malcolmst, 2020-10-16:
QQ - does the stop button work yet? I just tried to stop that simulation but it doesn't appear to be stopping. (The stop button is a great feature btw!)

acschang, 2020-10-16:
> QQ - does the stop button work yet? I just tried to stop that simulation but it doesn't appear to be stopping. (The stop button is a great feature btw!)

Try refreshing your browser after pressing the stop button.  I believe the page does not automatically refresh when the stop button is pressed.

malcolmst, 2020-10-16:
Thanks! It didn't take effect before for some reason but it did now. Will take a look at those logs.

malcolmst, 2020-10-16:
I don't actually see any evidence of a comms problem in `e5fcd870-64d5-43a1-8d7f-99ee372c838e`, probably those messages just weren't getting through due to congestion causing burst loss or being out of comms range. This is different from what I saw locally, where the neighbors function didn't even show any peers.

closed 2021-01-21 by nkoenig