osrf/subt #674: Simulation crash due to WebsocketServer

zwn, 2020-10-16:
In simulation `d7741e9d-b97c-4757-a918-d02a8a9736df` I see the following stacktrace in the simulation console:

<pre>
[Err] [WebsocketServer.cc:537] Null pointer to a conection. This should not happen.
Stack trace (most recent call last) in thread 66:
#7    Object "", at 0xffffffffffffffff, in 
#6    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fb409648a3e, in clone
#5    Object "/lib/x86_64-linux-gnu/libpthread.so.0", at 0x7fb40930f6da, in start_thread
#4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fb4065af6de, in std::error_code::default_error_condition() const
#3    Object "/usr/lib/x86_64-linux-gnu/libignition-transport7.so.7", at 0x7fb40501b863, in ignition::transport::v7::NodeSharedPrivate::PublishThread()
#2    Object "/usr/lib/x86_64-linux-gnu/libignition-transport7.so.7", at 0x7fb405038b7a, in ignition::transport::v7::RawSubscriptionHandler::RunRawCallback(char const*, unsigned long, ignition::transport::v7::MessageInfo const&)
#1    Object "/usr/lib/x86_64-linux-gnu/ign-launch-1/plugins/libignition-launch-websocket-server.so", at 0x7fb4054a1e9c, in ignition::launch::WebsocketServer::OnWebsocketSubscribedMessage(char const*, unsigned long, ignition::transport::v7::MessageInfo const&)
#0    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fb40659bfa3, in std::_Rb_tree_increment(std::_Rb_tree_node_base*)
Segmentation fault (Signal sent by the kernel [(nil)])
./run_sim.bash: line 17:    54 Segmentation fault      (core dumped) ign launch -v 4 $@
[Dbg] [VisibilityRfModel.cc:134] Range: 0, Exp: 1.5, Num hops: 0, TX: 20
</pre>

nkoenig, 2020-10-16:
Thanks. The fix is here: https://github.com/ignitionrobotics/ign-launch/pull/68

nkoenig, 2020-12-16:
This has been fixed and released.

closed 2020-12-16 by nkoenig