osrf/subt #675: Robots falling through the floor in different worlds

wolfgangschwab, 2020-10-17:
Sorry about the late reporting of this issue. We just noticed this behaviour only a two days ago and were too busy with the ending submission window to report this earlier. 

We noticed in about 2/3 of our runs that robots were falling through the floor. This happened in different worlds. 
We really would appreciate a fix of this issue before the cave circuit runs start. 

Here are some short videos of such fallings:
Cave Practice 1:
[image]

Cave Practice 3:
[image]

Cave Qualification:
[image]

Cave Simple 1:
[image]

Cave Simple 3:
[image]

These were all cloudsim runs. If you need more informations, I can provide them.

nkoenig, 2020-10-21:
This is happening because your solution is sending `nan` values to `cmd_vel` and physics is not able to handle this case. There is a patch to physics here: https://github.com/ignitionrobotics/ign-physics/pull/137. I also recommend modifying your code to eliminate the `nan` values.

We have seen the `nan` values in the ROS logs:
```
84.272000000 INFO /X2n2_robot_controller [/home/developer/workspaces/subt_seed_ws/src/subt_seed/src/robot_Controller.cpp:4832(setVelocties)] [topics: /rosout, /X2n2/cmd_vel, /X2n2/wand_visualization_marker, /X2n2/checkNodeList_req, /X2n2/checkNodeList_and_insert_req, /X2n2/checkEdgeList_req, /X2n2/checkEdgeList_and_insert_req, /X2n2/shortestPath_req, /X2n2/nearestEdge1End_req, /X2n2/artefaktmeldungen, /X2n2/breadcrumb/deploy, /statistics]   cmd_vel tatsaechliche Vorgabe: vx=1.349041, vy=0.000000, vz=0.000000, angz=nan
```

wolfgangschwab, 2020-10-27:
Hi Nate, 

thank you very much for the feedback. We did not expect this reason. We definitely will change your controller to avoid sending a nan. But thanks for fixing it in ign-physics. 

Wolfgang

closed 2020-10-27 by wolfgangschwab