osrf/subt #676: DNS issue with www.subtchallenge.com

malcolmst, 2020-10-17:
I thought this might have been just a temporary glitch, but I've noticed for quite a while that computers on my network cannot resolve www.subtchallenge.com using a default DNS configuration (my ISP is Spectrum). If I add `8.8.8.8` as a DNS server though, it resolves fine. On my phone, www.subtchallenge.com resolves fine on 4G but not wifi on my network. I believe my default DNS configuration is using a DNS service hosted by the cable modem on 127.0.0.53, but can try to get more details if it would be helpful.

I expect this is some issue with my ISP, but wanted to let you know in case it's more widespread than just me.

Before adding `8.8.8.8`:

```
nslookup www.subtchallenge.com
Server:		127.0.0.53
Address:	127.0.0.53#53

** server can't find www.subtchallenge.com: SERVFAIL
```

After:
```
nslookup www.subtchallenge.com
Server:		8.8.8.8
Address:	8.8.8.8#53

Non-authoritative answer:
Name:	www.subtchallenge.com
Address: 192.5.18.235
```

malcolmst, 2020-10-17:
Also, this is just www.subtchallenge.com, www.subtchallenge.world works fine.

zwn, 2020-10-19:
I am also using the cloudflare dns resolvers and it fails for me as well. I've created a ticket there just in case the problem is on cloudflare side: https://community.cloudflare.com/t/subtchallenge-com-not-resolving/214948

angelacmaio, 2020-10-19:
This information has been routed to the proper group. Since subtchallenge.com is not part of the SubT Virtual Testbed, I will close this issue. If you have any further information, please send to SubTChallenge@darpa.mil.

closed 2020-10-19 by angelacmaio