osrf/subt #677: Availability of Cloudsim Logs

AravindaDP, 2020-10-17:
Now that we have run dozens of cloud simulations I'm curious how long could we expect the logs will be retained online if we needed to retrieve them back in future.

Will they available indefinitely (Or at least until conclusion of Final event next year?) No matter how many simulations we have?
Or should we make a backup immediately in case cloud sim runs out of storage space and begin to drop off older simulation logs?

On the other hand there are simulations that we run to troubleshoot and no longer interested. So may be an option to delete simulations by competitors themselves? (Obviously it might need additional check to prevent accidentially deleting competition runs if they are available in same fashion)

AravindaDP, 2020-11-19:
Also is there a command line url pattern we could use to download these logs either using wget or aws cli? (Similar to how we get circuit event logs)

nkoenig, 2020-12-17:
We recommend posting this competition-related question on the [SubT Challenge Community Forum](https://community.subtchallenge.com/login) so it can be addressed by DARPA.

closed 2020-12-17 by nkoenig