osrf/subt #684: What versions of system components are going to be used for the finals?

zwn, 2020-10-27:
Are there going to be any updates of the components between now and the finals? I am asking mainly about the major components like ignition (blueprint, citadel, dome or edifice?), ros (melodic or noetic?) and ubuntu (bionic or focal?). There no plans to utilize ros2, right?

If there are going to be upgrades, can they be done as soon as possible?

pauljurczak, 2020-11-20:
ROS2 and Ubuntu 20.04 would be very nice indeed. I had a lot of issues with using the newest compilers and libraries on Ubuntu 18.04 and Melodic. https://github.com/osrf/subt/wiki/System%20Requirements states: *ROS Melodic is used for all SubT Circuits and the final*. I hope this is not set in stone.

peci1, 2020-12-05:
I understand the sentence from the wiki quite clearly. And I consider it a good idea to not exchange the underlying OS and switch to a version of ROS that doesn't work with Python 2. Let's just finish with what we have and have an upgrade party after the finals :)

However, Ignition version is not mentioned in the wiki. I see some development here on citadel branches. Does that mean finals will run on citadel? Please, we need to know, the earlier, the better. I guess there could be some difficulties with Citadel on Ubuntu Bionic, as it requires a newer gcc version that Ubuntu ships by default (at least the CI script installs gcc8).

pauljurczak, 2020-12-13:
Just to add an extra perspective to this topic: NASA Space Robotics Challenge Phase 2, which is a similar simulation competition, will upgrade their software platform for 2021 finals to Gazebo 11 and ROS Noetic.

zwn, 2020-12-15:
I am not sure how to interpret the fact that this question has gone for so long without an answer. Is it because it is not yet known what versions will be used? Can we know at least what the possibilities under consideration are? I see stuff being merged into "dome" - so is "dome" being considered? What is the reason for the silence? Whose help is being wanted here? I am confused.

pauljurczak, 2020-12-15:
My guess is that they are busy with final touches on final release. SubT v5.0 Final Event software release was scheduled for Nov. 2020, so we will probably find out soon.

angelacmaio, 2020-12-16:
Indeed, regression testing and new docker images have been in the works and are now released. Please see the updated [Release Notes](https://github.com/osrf/subt/wiki/release_notes) and installation information in the [Get Started page](https://github.com/osrf/subt/wiki/Get%20Started).

closed 2020-12-16 by angelacmaio

zwn, 2020-12-17:
For anyone looking for the actual statement that the current versions will be used for the final event - [it is in the forums](https://community.subtchallenge.com/t/subt-simulator-update/1489/1).