osrf/subt #685: Deploying breadcrumbs should be done through ROS service, not topic

zwn, 2020-10-27:
Currently we are expected to [publish on a ros topic an empty message](https://github.com/osrf/subt/wiki/api#breadcrumbs). But publishing on a topic is inherently unreliable means of communication - dropping a message is just a debug level event in roscpp - so topics are useful mainly for periodic sending of state information but really unsuitable for sending state changes.

Please change the API to service to remove one bit of unnecessary unreliability from the system.

wolfgangschwab, 2020-10-27:
I like to deploy breadcrumbs using a topic. A ROS service is a synchronous service and has the advantage that you get an answer immediately. But you have to wait for the answer. This might take a while and the process waits for the answer and cannot generate or send other commands meanwhile. Additionally I would not expect an important answer or feedback from this service as the breadcrumb will be deployed anyway until none is available any more. And also we should not have such communication problems here, that we have in the communication between the robots. 

So I would stay with a topic to deploy the breadcrumbs.

zwn, 2020-10-28:
@wolfgangschwab Messages on topic can get dropped - by design (checkout roscomm sources as I have). This fact is unrelated to any problems there might be on the cloudsim side. This is a design decision inside roscomm library that is being used to implement both parties. If the message gets dropped (due to CPU pressure for example), the breadcrumb is not deployed. Additionally, we do no have any direct feedback that this has happened, so there is not a way to recover. Not sending state changes over unreliable channel without any means for correction is communication 101. Many other things are services like getting position wrt artifact origin or the insides of CommsClient and there are ways to do so asynchronously while there is no way to reliably drop a breadcrumb.

zwn, 2020-11-09:
There is the same problem with the marsupials detachment https://github.com/osrf/subt/wiki/api#marsupials. Here a possible workaround is to keep publishing because there is only one state change. But still, that is sub-optimal.