osrf/subt #686: Crash when exiting gui

zwn, 2020-10-29:
I run the simulation headless by default. While trying to launch the gui independently, I got segmentation fault on exit.

Steps I have taken:
- start headless simulation with one robot using the simulation docker image
- docker exec another bash inside the simulation container
- `mkdir ~/.ignition/gazebo` because without it the gui fails to start claiming it cannot copy config file
- `ign gazebo -g` to launch the gui
- move the camera around the starting zone
- select Quit from the menu

This is the content of the terminal:

<pre>$ ign gazebo -g
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to &apos;/tmp/runtime-developer&apos;
[BUG] Segmentation fault at 0x00007f70a9763198
ruby 2.5.1p57 (2018-03-29 revision 63029) [x86_64-linux-gnu]

-- Machine register context ------------------------------------------------
 RIP: 0x00007f70f32b5cd7 RBP: 0x000055edc44ab820 RSP: 0x00007fffde4a1020
 RAX: 0x00007f70a9763198 RBX: 0x000055edc44ab820 RCX: 0x0000000000000000
 RDX: 0x000055edc44ab820 RDI: 0x000055edc44ab810 RSI: 0x000055edc3f10b20
  R8: 0x0000000000000099  R9: 0x00007f70d403bb90 R10: 0x0000000000000065
 R11: 0x00007f70f48fbbd0 R12: 0x000055edc44ab810 R13: 0x0000000000000001
 R14: 0x0000000000000001 R15: 0x00007f70f33afcb0 EFL: 0x0000000000010213

-- C level backtrace information -------------------------------------------
/usr/lib/x86_64-linux-gnu/libruby-2.5.so.2.5(0x7f70f8182aa5) [0x7f70f8182aa5]
/usr/lib/x86_64-linux-gnu/libruby-2.5.so.2.5(0x7f70f8182cdc) [0x7f70f8182cdc]
/usr/lib/x86_64-linux-gnu/libruby-2.5.so.2.5(0x7f70f804c884) [0x7f70f804c884]
/usr/lib/x86_64-linux-gnu/libruby-2.5.so.2.5(0x7f70f8112ab2) [0x7f70f8112ab2]
/lib/x86_64-linux-gnu/libc.so.6(0x7f70f7c12fd0) [0x7f70f7c12fd0]
/usr/lib/x86_64-linux-gnu/libQt5Qml.so.5(0x7f70f32b5cd7) [0x7f70f32b5cd7]
/usr/lib/x86_64-linux-gnu/libQt5Qml.so.5(_ZN17QQmlPropertyCacheD1Ev+0x6e) [0x7f70f34abfce]
/usr/lib/x86_64-linux-gnu/libQt5Qml.so.5(_ZN17QQmlPropertyCacheD0Ev+0x9) [0x7f70f34ac419]
/usr/lib/x86_64-linux-gnu/libQt5Qml.so.5(0x7f70f3478462) [0x7f70f3478462]
/usr/lib/x86_64-linux-gnu/libQt5Qml.so.5(0x7f70f3478489) [0x7f70f3478489]
/lib/x86_64-linux-gnu/libc.so.6(0x7f70f7c170f1) [0x7f70f7c170f1]
/lib/x86_64-linux-gnu/libc.so.6(0x431ea) [0x7f70f7c171ea]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xee) [0x7f70f7bf5b9e]
ruby(_start+0x2a) [0x55edc10208fa]

-- Other runtime information -----------------------------------------------

* Process memory map:

55edc1020000-55edc1021000 r-xp 00000000 103:01 15760677                  /usr/bin/ruby2.5
55edc1220000-55edc1221000 r--p 00000000 103:01 15760677                  /usr/bin/ruby2.5
55edc1221000-55edc1222000 rw-p 00001000 103:01 15760677                  /usr/bin/ruby2.5
55edc169f000-55edc6a23000 rw-p 00000000 00:00 0                          [heap]
7f7044000000-7f70478b9000 rw-p 00000000 00:00 0 
7f70478b9000-7f7048000000 ---p 00000000 00:00 0 
7f7048000000-7f704a000000 rw-s 00000000 00:06 497                        /dev/nvidiactl
7f704a000000-7f704c000000 rw-s 00000000 00:06 497                        /dev/nvidiactl
7f704c000000-7f704c021000 rw-p 00000000 00:00 0 
7f704c021000-7f7050000000 ---p 00000000 00:00 0 
7f7050000000-7f7050021000 rw-p 00000000 00:00 0 
7f7050021000-7f7054000000 ---p 00000000 00:00 0 
7f7054000000-7f7054021000 rw-p 00000000 00:00 0 
7f7054021000-7f7058000000 ---p 00000000 00:00 0 
7f705c000000-7f705c021000 rw-p 00000000 00:00 0 
7f705c021000-7f7060000000 ---p 00000000 00:00 0 
7f7060000000-7f7060021000 rw-p 00000000 00:00 0 
7f7060021000-7f7064000000 ---p 00000000 00:00 0 
7f7064000000-7f7064021000 rw-p 00000000 00:00 0 
7f7064021000-7f7068000000 ---p 00000000 00:00 0 
7f7068000000-7f7068021000 rw-p 00000000 00:00 0 
7f7068021000-7f706c000000 ---p 00000000 00:00 0 
7f706c000000-7f706c021000 rw-p 00000000 00:00 0 
7f706c021000-7f7070000000 ---p 00000000 00:00 0 
7f7074000000-7f7077ff8000 rw-p 00000000 00:00 0 
[... 1449 lines cut ...]
7f70f8675000-7f70f8679000 r--s 00000000 103:01 13511806                  /var/cache/fontconfig/de156ccd2eddbdc19d37a45b8b2aac9c-le64.cache-7
7f70f8679000-7f70f867d000 r--s 00000000 103:01 13511803                  /var/cache/fontconfig/c57959a16110560c8d0fcea73374aeeb-le64.cache-7
7f70f867d000-7f70f8683000 r--s 00000000 103:01 13511795                  /var/cache/fontconfig/3047814df9a2f067bd2d96a2b9c36e5a-le64.cache-7
7f70f8683000-7f70f8687000 r--s 00000000 103:01 13511805                  /var/cache/fontconfig/d589a48862398ed80a3d6066f4f56f4c-le64.cache-7
7f70f8687000-7f70f868c000 r--s 00000000 103:01 11545786                  /home/developer/.cache/ign-gazebo-gui/qmlcache/5b2121aef63192755e7fa30131090501014c6cc0.qmlc
7f70f868c000-7f70f8694000 r-xs 00005000 103:01 11545786                  /home/developer/.cache/ign-gazebo-gui/qmlcache/5b2121aef63192755e7fa30131090501014c6cc0.qmlc
7f70f8694000-7f70f8695000 rw-p 00000000 00:00 0 
7f70f8695000-7f70f8696000 r-xp 00000000 00:00 0 
7f70f8699000-7f70f869b000 r--s 00000000 103:01 15760677                  /usr/bin/ruby2.5
7f70f869b000-7f70f869d000 rw-p 00000000 00:00 0 
7f70f869d000-7f70f869e000 r--p 00027000 103:01 15760490                  /lib/x86_64-linux-gnu/ld-2.27.so
7f70f869e000-7f70f869f000 rw-p 00028000 103:01 15760490                  /lib/x86_64-linux-gnu/ld-2.27.so
7f70f869f000-7f70f86a0000 rw-p 00000000 00:00 0 
7fffddca5000-7fffde4a4000 rw-p 00000000 00:00 0                          [stack]
7fffde4f0000-7fffde4f3000 r--p 00000000 00:00 0                          [vvar]
7fffde4f3000-7fffde4f5000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]

[NOTE]
You may have encountered a bug in the Ruby interpreter or extension libraries.
Bug reports are welcome.
For details: http://www.ruby-lang.org/bugreport.html

Aborted (core dumped)
</pre>

zwn, 2020-10-29:
When trying log playback, I got again a crash when quitting the gui window:

<pre>$ ign gazebo -r --playback /tmp
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to &apos;/tmp/runtime-developer&apos;
[BUG] Segmentation fault at 0x00007fbfbf0f6198
ruby 2.5.1p57 (2018-03-29 revision 63029) [x86_64-linux-gnu]

-- Machine register context ------------------------------------------------
 RIP: 0x00007fc0088d0cd7 RBP: 0x000055b1e3233440 RSP: 0x00007ffef1bf5f30
 RAX: 0x00007fbfbf0f6198 RBX: 0x000055b1e3233440 RCX: 0x0000000000000000
 RDX: 0x000055b1e3233440 RDI: 0x000055b1e3233430 RSI: 0x000055b1e2915a10
  R8: 0x00000000000000d9  R9: 0x00007fbfe803bb90 R10: 0x0000000000000065
 R11: 0x00007fc009f16bd0 R12: 0x000055b1e3233430 R13: 0x0000000000000001
 R14: 0x0000000000000001 R15: 0x00007fc0089cacb0 EFL: 0x0000000000010213

-- C level backtrace information -------------------------------------------
/usr/lib/x86_64-linux-gnu/libruby-2.5.so.2.5(0x7fc00d79daa5) [0x7fc00d79daa5]
/usr/lib/x86_64-linux-gnu/libruby-2.5.so.2.5(0x7fc00d79dcdc) [0x7fc00d79dcdc]
/usr/lib/x86_64-linux-gnu/libruby-2.5.so.2.5(0x7fc00d667884) [0x7fc00d667884]
/usr/lib/x86_64-linux-gnu/libruby-2.5.so.2.5(0x7fc00d72dab2) [0x7fc00d72dab2]
/lib/x86_64-linux-gnu/libc.so.6(0x7fc00d22dfd0) [0x7fc00d22dfd0]
/usr/lib/x86_64-linux-gnu/libQt5Qml.so.5(0x7fc0088d0cd7) [0x7fc0088d0cd7]
/usr/lib/x86_64-linux-gnu/libQt5Qml.so.5(_ZN17QQmlPropertyCacheD1Ev+0x6e) [0x7fc008ac6fce]
/usr/lib/x86_64-linux-gnu/libQt5Qml.so.5(_ZN17QQmlPropertyCacheD0Ev+0x9) [0x7fc008ac7419]
/usr/lib/x86_64-linux-gnu/libQt5Qml.so.5(0x7fc008a93462) [0x7fc008a93462]
/usr/lib/x86_64-linux-gnu/libQt5Qml.so.5(0x7fc008a93489) [0x7fc008a93489]
/lib/x86_64-linux-gnu/libc.so.6(0x7fc00d2320f1) [0x7fc00d2320f1]
/lib/x86_64-linux-gnu/libc.so.6(0x431ea) [0x7fc00d2321ea]
/usr/lib/x86_64-linux-gnu/libruby-2.5.so.2.5(ruby_run_node+0x10) [0x7fc00d66f410]
/usr/lib/x86_64-linux-gnu/libruby-2.5.so.2.5(0x7fc00d6f3a07) [0x7fc00d6f3a07]
/usr/lib/x86_64-linux-gnu/libruby-2.5.so.2.5(0x7fc00d7863c9) [0x7fc00d7863c9]
/usr/lib/x86_64-linux-gnu/libruby-2.5.so.2.5(0x7fc00d7948f3) [0x7fc00d7948f3]
/usr/lib/x86_64-linux-gnu/libruby-2.5.so.2.5(0x7fc00d78ad85) [0x7fc00d78ad85]
/usr/lib/x86_64-linux-gnu/libruby-2.5.so.2.5(0x7fc00d790b64) [0x7fc00d790b64]
/usr/lib/x86_64-linux-gnu/libruby-2.5.so.2.5(0x7fc00d66b0c4) [0x7fc00d66b0c4]
/usr/lib/x86_64-linux-gnu/libruby-2.5.so.2.5(ruby_exec_node+0x1d) [0x7fc00d66cf4d]
/usr/lib/x86_64-linux-gnu/libruby-2.5.so.2.5(ruby_run_node+0x1e) [0x7fc00d66f42e]
ign gazebo gui(0x55b1df4318cb) [0x55b1df4318cb]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xe7) [0x7fc00d210b97]
ign gazebo gui(_start+0x2a) [0x55b1df4318fa]

</pre>

nkoenig, 2021-01-13:
Do you still see this issue with Ignition Dome?

closed 2021-02-01 by nkoenig