osrf/subt #687: Suggestion: Bluetooth/Audio sensor

AravindaDP, 2020-10-29:
It would be nice if robots could have bluetooth/audio sensors as in systems track for the finals event.

https://community.subtchallenge.com/t/suitability-of-existing-virtual-testbed-vehicles-for-missions/1221
https://community.subtchallenge.com/t/cell-phone-bluetooth-discoverable/833
https://community.subtchallenge.com/t/phone-artifact-bluetooth/1213

I guess both of these type sensors may not available in Gazebo Ignition so might need custom sensor development like in gas sensor plugin. Probably easiest is to implement a bluetooth sensor similar to gas sensor but instead of Boolean true false state this could give rssi estimation. (Do we need to model accurate rssi like in radio visibility model or simple equation based on distance?)

AravindaDP, 2020-11-01:
Probably better implementation for bluetooth could have if wireless reciever/transmitter was implemented on ignition gazebo as proposed here
https://github.com/ignitionrobotics/ign-sensors/issues/28

peci1, 2021-01-19:
The logical audio plugin has been merged in ign-gazebo3 (ignitionrobotics/ign-gazebo#401).

AravindaDP, 2021-01-19:
I can see recent release in dome as well so assume this got merged there as well. It would be very interesting to have multimodal sensing options in virtual challenge as well. I haven't checked but I guess thermal camera also completed/near completion?
Something like below would be very interesting.

Artifact_type|Visual|Audio|Thermal
---|---|---|---
Rescue_Randy|x|x|x
Vent|x| |x
Cellphone|x|x| 
Backpack|x| | 

I guess we could afford audio sensor on all models may be? (Or only on selected models would also be fine)
Having thermal camera on only custom configurations with subject to increase in cost is expected.