osrf/subt #689: Playback of logs containing Freyja config 2 - upper/lower case problem

zwn, 2020-10-29:
When I try to play back state.tlog of our run with ROBOTIKA_FREYJA_SENSOR_CONFIG_2 following the instructions at 
https://github.com/osrf/subt/wiki/Cloudsim-Log-Playback I get only partial success. It runs but the wheel meshes are not loaded. This is the error message I get on the terminal:

```
[GUI] [Err] [SystemPaths.cc:459] File [/home/developer/.ignition/fuel/fuel.ignitionrobotics.org/OpenRobotics/models/ROBOTIKA_FREYJA_SENSOR_CONFIG_2/3/meshes/wheel_left.dae] resolved to path [/home/developer/.ignition/fuel/fuel.ignitionrobotics.org/OpenRobotics/models/ROBOTIKA_FREYJA_SENSOR_CONFIG_2/3/meshes/wheel_left.dae] but the path does not exist
```

The path indeed does not exist. But similar path with lowercase config exists. When switch to directory `/home/developer/.ignition/fuel/fuel.ignitionrobotics.org/OpenRobotics/models/` and execute:
```
ln -s Robotika_Freyja_Sensor_Config_2  ROBOTIKA_FREYJA_SENSOR_CONFIG_2
```
the playback starts working.

nkoenig, 2020-12-08:
This issue is resolved here https://github.com/ignitionrobotics/ign-fuel-tools/pull/130.

nkoenig, 2020-12-16:
This has been fixed and released in ign-fuel-tools.

closed 2020-12-16 by nkoenig