osrf/subt #691: Horizontal resolution (samples) of 3D lidar are way too higher than real sensors.

AravindaDP, 2020-10-31:
As pointed in https://github.com/osrf/subt/issues/680#issuecomment-714930086 horizontal resolution of almost all 3D lidars are way too higher than real sensor (Assuming VLP-16)

It should be 1800 for 10Hz and 1200 if running at 15Hz. http://www.mapix.com/wp-content/uploads/2018/07/63-9229_Rev-H_Puck-_Datasheet_Web-1.pdf

I have being suggesting incoming PR to submitted models to correct this along with already submitted models but I couldn't figure out where corresponding configuration resides for default robots (X1 and X2). I see a default definition at https://github.com/osrf/subt/blob/master/subt_sensors/urdf/3d_lidar.urdf.xacro#L40 but somehow it get overridden somewhere else (for example X1_SENSOR_CONFIG_3)

On a side note may be it's good idea to increase resolution (number of actual measurements simulated vs interpolated) now that we reduce horizontal samples?

zwn, 2020-11-02:
> On a side note may be it's good idea to increase resolution (number of actual measurements simulated vs interpolated) now that we reduce horizontal samples?

I was just about to create an issue for this :+1:. Sending interpolated data brings no new information. It only stresses the communication infrastructure, logging infrastructure etc. So if 1800 is not possible wrt performance, just lower the actual number of samples to a level that is reasonable but don't add any interpolated data. If 1000 is the number than be it and set resolution to 1. Thanks.

peci1, 2021-01-19:
#753 addresses this issue.

closed 2021-01-22 by acschang