osrf/subt #692: Setting boost library version in cmake configuration

pauljurczak, 2020-11-03:
It would help me to have an option to set boost library to a specific version in cmake configuration. Catkin build works fine with boost 1.65, but some libraries I'm using require a newer version. Catkin build failed with boost v1.72, see: #444. I couldn't find any option to control, which boost version is used for catkin build. Can we get one?

mjcarroll, 2020-11-16:
You should be able to search for a specific version using standard cmake macros.

You can do `find_package(Boost 1.72 REQUIRED COMPONENTS filesystem ... etc)`

There is a chance for this to cause issues, though, as many pieces of ROS melodic may be linked to the older (system-default) version of boost.

mjcarroll, 2021-01-08:
@pauljurczak Did the `find_package` resolve the issue for you?

pauljurczak, 2021-01-09:
I meant setting boost version in `subt_ws` make files. It is no longer an issue for my project, since I'm using header only boost libraries installed in user space.

closed 2021-01-09 by pauljurczak