osrf/subt #697: How to find name of marsupial child?

m3d, 2020-11-07:
Hello,
is there a way how to find the name of marsupial child? (it is needed for `detach` call)
We use variable names for strategy encoding and so this link is not clear from the robot's perspective.
thanks a lot
   Martin
p.s. or we would have to encode it in parents name
https://github.com/osrf/subt/wiki/api#marsupials

m3d, 2020-11-07:
One possibility is to detect it from shared 3D position via radio - if there is some other robot within small radius on the start then it is maybe marsupial child/parent ...

nkoenig, 2020-11-09:
You have a couple options:

1. Bake the child robot's name into your code.
2. Use the communication model.

m3d, 2020-11-16:
It looks like that the marsupial drone can detach itself:
```
71.784000000 INFO /wait [wait.py:69(check_marsupials)] [topics: /clock, /rosout] Marsupial: /A100L/detach 
```
so it is not necessary to communicate with the parent "how to detach". OTOH on CloudSim I am not able to run it (it looks like the simulation is terminated after detach call). Is anybody using marsupials on CloudSim? Can you please confirm that it works for you?
thanks
    Martin

p.s. @nkoenig - thanks for info ... so parent does not know directly that it is carrying a child ... both hints (communicate it and extend name) works for us

m3d, 2020-11-18:
The problem was on our side (missing one publishing module present for UGVs/breadcrumbs but missing for UAVs). So the drone finally took off: https://youtu.be/SnDr2xbqlvE
thanks
m.

closed 2020-11-18 by m3d