osrf/subt #698: CloudSim visualization menu of marsupial robots

m3d, 2020-11-07:
Minor detail, that when we deploy marsupial robots (Explorer R2 + SSCI X4) the menu does not expand. When you type some letters then it is updated., but the "Robots" group is still missing:
[image]

The robots names are `A100L` and `B100L`.

vinhanguyen, 2020-11-20:
Fixed in https://github.com/osrf/subtportal/pull/165

m3d, 2020-11-22:
> 
> 
> Fixed in [osrf/subtportal#165](https://github.com/osrf/subtportal/pull/165)

Thanks - is the "osrf/subtportal" repository publicly available? I get 404 when I click on the link.

cameron-ridgewell, 2020-11-22:
> Thanks - is the "osrf/subtportal" repository publicly available? I get 404 when I click on the link.

That repository is private

angelacmaio, 2020-12-17:
Fixed in osrf/subtportal#165

closed 2020-12-17 by angelacmaio