osrf/subt #707: Can't download competition logs

peci1, 2020-11-17:
I received the email with link to download competition logs, which mentions the command 

    aws s3 sync s3://web-cloudsim-production-logs/gz-logs/CTU-CRAS-NORLAB/circuit_logs/cave/

I tried running it both with aws CLI 1 and 2 on machines with credentials set up (files in `~/.aws` exist). But with no success. I always get 

    An error occurred (AccessDenied) when calling the ListObjectsV2 operation: Access Denied

Is there something else I need to do prior to calling this command?

Commands `aws configure get aws_access_key_id` and `rn3NoW7NSdij25O1nLkFInM/MuL/47+O9iffR08G` both return nonempty strings.

zwn, 2020-11-17:
I am downloading ok (team robotika).

peci1, 2020-11-17:
Now this is interesting. Calling the command with lowercase team name doesn't give an error, but points to an empty directory:

    $ aws s3 ls s3://web-cloudsim-production-logs/gz-logs/ctu-cras-norlab/circuit_logs/cave/
    $ aws s3 ls s3://web-cloudsim-production-logs/gz-logs/CTU-CRAS-NORLAB/circuit_logs/cave/

    An error occurred (AccessDenied) when calling the ListObjectsV2 operation: Access Denied
    $

peci1, 2020-11-17:
Could it be related to #629? I know @nkoenig told that organization name is case-insensitive, but I still think we proved otherwise. But in #629, uppercase name was the right form :/

nkoenig, 2020-11-17:
The problem was with the script to update IAM rules. Capitalization of `CTU-CRAS-NORLAB` was not preserved during the update. 

Try again please.

peci1, 2020-11-18:
Now it works, thanks!

closed 2020-11-18 by peci1