osrf/subt #709: CloudSim visualization - highlight multiple objects

m3d, 2020-11-20:
Hello,
I am using CloudSim visualization to see how are multiple robots doing and one feature, which helps me to see "overview" is `select` object (instead of `move  to` or `follow`). Would it be possible to highlight multiple objects? Maybe not only robots but also artifacts of given type?
[image]

(this is just "nice to have" feature)

m3d, 2020-11-20:
p.s. I highlighted the selected robot in the case you do not see it on the image above:

[image]

AravindaDP, 2020-11-21:
While we are at it it might be more beneficial if highlight is more prominent (Like a call out with entity name that is scale invariant) it's hard to spot an entity like robot/artifact in a zoomed out view such as above)

Alternatively if we can have something like mini-map view used in cave circuit showcase I think that may be more beneficial.

In any case that is again just a nice to have from my point of view as well.