osrf/subt #711: Permutation table for competition Cave Worlds?

m3d, 2020-11-23:
Hello,
is there somewhere table of which world corresponds to which in the [Showcase](https://www.youtube.com/watch?v=wv-nXkhr5Sk)? Thanks to @zwn for discovering this issue, that CloudSim `Cave Cirtuit 1` is actually `World 3` in the competition (Robotika got 5, 2, 5 points). What are the others?
thanks
  Martin

m3d, 2020-11-23:
p.s. `Cave Circuit World 8` is `World 8` ;)

AravindaDP, 2020-11-24:
It's in the competition logs email.

```
align to the following world names as described in the Cave Circuit Virtual Competition Showcase:

World 1: Fork in the Road
World 2: Glyph
World 3: T-Rex Handstand
World 4: The High Road
World 5: Uphill Battle
World 6: Five Fingers
World 7: Triple Decker
World 8: Rock and a Hard Place
```

m3d, 2020-11-24:
You are absolutely, right! _The worlds are also defined by a different number order than those used in the Showcase. Please refer to the worlds as listed on the SubT Tech Repo when reviewing your logs_ ... so now just to match numbers and names
thanks for hint

peci1, 2020-11-24:
cave_circuit_1 (Fork in the road)
cave_circuit_2 (Glyph)
cave_circuit_3 (T-Rex Handstand)
cave_circuit_4 (The high road)
cave_circuit_5 (Uphill Battle)
cave_circuit_6 (Five Fingers)
cave_circuit_7 (Triple Decker)
cave_circuit_8 (Rock and a Hard Place; Jeonlan cave)

m3d, 2020-11-24:
Thanks, closing this "issue"

closed 2020-11-24 by m3d

m3d, 2020-11-24:
p.s. just for reference attached picture of presented worlds:
[image]