osrf/subt #712: Exception in simulator log

peci1, 2020-11-24:
I found this in ros/UUID/master.log  in the sim container logs:

```
[rosmaster.master][INFO] 2020-10-17 19:54:16,768: publisherUpdate[/subt/kinematic_states/uav5] -> http://sim-7e2b5148-6635-4f7a-88fb-70fd097cfc1c-c-1-gzserver:33041/ []
[rosmaster.master][INFO] 2020-10-17 19:54:16,769: publisherUpdate[/subt/kinematic_states/uav5] -> http://sim-7e2b5148-6635-4f7a-88fb-70fd097cfc1c-c-1-gzserver:33041/ []: sec=0.00, exception=<Fault -1: 'publisherUpdate: unknown method name'>
[rosmaster.threadpool][ERROR] 2020-10-17 19:54:16,769: Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosmaster/threadpool.py", line 218, in run
    result = cmd(*args)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosmaster/master_api.py", line 210, in publisher_update_task
    ret = xmlrpcapi(api).publisherUpdate('/master', topic, pub_uris)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1243, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1602, in __request
    verbose=self.__verbose
  File "/usr/lib/python2.7/xmlrpclib.py", line 1283, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1316, in single_request
    return self.parse_response(response)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1493, in parse_response
    return u.close()
  File "/usr/lib/python2.7/xmlrpclib.py", line 800, in close
    raise Fault(**self._stack[0])
Fault: <Fault -1: 'publisherUpdate: unknown method name'>
```

This exception repeats about 10-20 times. According to timestamps, it seems like it happened on the very end (teardown) of the simulation. So probably nothing serious, but I just wanted to let you know the error is there.

nkoenig, 2021-01-13:
This issue seems to be harmless. See https://github.com/ros/ros_comm/issues/572, which seems to indicate that it's a race condition in ROS during shutdown.

closed 2021-01-13 by nkoenig

peci1, 2021-01-13:
Thanks for checking it.