osrf/subt #716: Back wall of staging area in Tunnel Circuit Worlds

m3d, 2020-11-28:
Hello,
can you please add extra wall for back side of the tunnel staging area? There is invisible barrier for the drones resulting in crash:
[image]

I know that it would be better not to return to the staging area, but it happens, sometimes.
thank you
   Martin/Robotika Team

adlarkin, 2021-01-14:
I have update the tunnel staging area to include a back wall. @m3d, would you mind testing this and confirming that things work for you? You'll probably need to re-download the latest tunnel staging area model from fuel. Here are the steps for re-downloading the most up-to-date model:

1. Delete the current subt tunnel staging area model from your fuel cache, if you have one:
```
rm -rf ~/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/subt_tunnel_staging_area/
```
2. Download the newest model:
```
ign fuel download -u "https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/subt_tunnel_staging_area"
```
3. Launch a tunnel circuit world:
```
ign launch tunnel_circuit_practice.ign worldName:=tunnel_circuit_practice_01 robotName1:=X1 robotConfig1:=X2_SENSOR_CONFIG_1
```

You should see something like this: 
[image]

m3d, 2021-01-14:
@adlarkin thanks for update - is it possible to test it as part of docker?

adlarkin, 2021-01-15:
> is it possible to test it as part of docker?

Yeah, but we will have to make a new release for this change to be incorporated. I believe we are waiting for a few other things (like model submissions) to be completed before making the next release. Once the next release is out with this change, I'll let you know!

adlarkin, 2021-01-22:
@m3d a new docker release has been made that includes this change. You can run the following commands to test:

```
$ docker pull osrf/subt-virtual-testbed:latest

$ wget https://raw.githubusercontent.com/osrf/subt/master/docker/run.bash

$ chmod +x run.bash

$ ./run.bash osrf/subt-virtual-testbed:latest competition.ign worldName:=tunnel_circuit_practice_01 circuit:=tunnel robotName1:=X1 robotConfig1:=X1_SENSOR_CONFIG_1
```

Let me know if you have any problems - if not, I'll close this issue.

adlarkin, 2021-02-01:
Closing this because the back wall has been appearing consistently for me with the new docker images

closed 2021-02-01 by adlarkin