osrf/subt #719: Very unrealistic physics during competition run for a falling UGV

peci1, 2020-12-10:
This happened to the poor X2 in competition run 07 of CTU-CRAS-NORLAB:

[image]

Full resolution video here: https://vimeo.com/489336330 .

It seems some highly unrealistic forces were generated during the fall.

rsawtell, 2021-05-26:
This appears to be very common now, we can reproduce it almost every time with the stairs in urban environment.

https://user-images.githubusercontent.com/700223/119652256-d8fb9100-bdf3-11eb-88b1-fdfa5b7a08d9.mp4

peci1, 2021-05-26:
@rsawtell I'd check whether the inertial properties of all links are reasonable. Especially, whether the wheels have some non-negligible weight.

rsawtell, 2021-05-26:
Looking at the CoRo_Allie sdf the wheels all have <mass>5.124</mass> and the body has <mass>26.6</mass>.

peci1, 2021-05-26:
Okay, the inertials look good. The collision element look good, too. This is becoming strange. What was happening before? Did the robot roll down another flight of stairs before it came to the view?

We do not observe this kind of problems in any finals world with ctu_cras_norlab robots. However, we do not roll down stairs :)

peci1, 2021-05-26:
The very newest version of gazebo (4.9.1) added a plugin for visualizing contacts. Can you use this plugin to visualize them to see whether there are any false contact points detected? You'll find the plugin in the three dot menu of the GUI...

nkoenig, 2021-06-21:
We have updated DART to resolve this issue in this [PR](https://github.com/ignition-forks/dart/pull/28). Let me know if you still see this problem.

closed 2021-07-01 by nkoenig