osrf/subt #728: Dome installation with catkin workspace setup does not compile

Samu24, 2020-12-17:
After following the instructions for setup with catkin workspace, the initial attempt to `catkin_make install` failed.

Branch of `osrf/subt` is set to `master` and is at commit `5fb318068651f9bf782c168eb176cdb80d444c04`

Terminal Log:

```commandline
Base path: /home/zsamuel/subt_ws
Source space: /home/zsamuel/subt_ws/src
Build space: /home/zsamuel/subt_ws/build
Devel space: /home/zsamuel/subt_ws/devel
Install space: /home/zsamuel/subt_ws/install
Creating symlink "/home/zsamuel/subt_ws/src/CMakeLists.txt" pointing to "/opt/ros/melodic/share/catkin/cmake/toplevel.cmake"
####
#### Running command: "cmake /home/zsamuel/subt_ws/src -DCATKIN_DEVEL_PREFIX=/home/zsamuel/subt_ws/devel -DCMAKE_INSTALL_PREFIX=/home/zsamuel/subt_ws/install -G Unix Makefiles" in "/home/zsamuel/subt_ws/build"
####
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/zsamuel/subt_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/melodic
-- This workspace overlays: /opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/zsamuel/subt_ws/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.29
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
CMake Error at /home/zsamuel/subt_ws/build/catkin_generated/order_packages.cmake:2 (message):
  Error(s) in package '/opt/ros/melodic/share/ros_ign_image/package.xml':

  Error(s):

  - The generic dependency on 'ignition-msgs6' is redundant with:
  build_depend, build_export_depend, exec_depend

  - The generic dependency on 'ignition-transport9' is redundant with:
  build_depend, build_export_depend, exec_depend
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/catkin_workspace.cmake:42 (include)
  CMakeLists.txt:69 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/home/zsamuel/subt_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/zsamuel/subt_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```

Samu24, 2020-12-17:
fyi @marco-tranzatto

nkoenig, 2020-12-17:
What version of ros-ign do you have?

```
dpkg -l | grep ros-ign
```

You should have 0.9.4. If you don't, then you could try

```
sudo apt-get update
sudo apt-get upgrade ros-melodic-ros-ign
```

iche033, 2020-12-18:
I ran into this issue too. Among other things I tried, I think updating catkin was the one that resolved the issue for me.

Samu24, 2021-01-04:
@nkoenig I have version `0.9.4-2bionic` on all packages found.

peci1, 2021-01-06:
I updated my catkin installation without a glitch of this type.

azeey, 2021-01-08:
It could be related to these lines: https://github.com/ignitionrobotics/ros_ign/blob/melodic/ros_ign_image/package.xml#L17-L18
``` xml
<depend condition="$IGNITION_VERSION == ''">ignition-msgs6</depend>
<depend condition="$IGNITION_VERSION == ''">ignition-transport9</depend>
```
The support for quoted values in the `condition` attribute was added in https://github.com/ros-infrastructure/catkin_pkg/pull/295, which I believe was released in the 0.4.23 version of `catkin_pkg` (python-catkin-pkg). Can you check what version you have? (`apt-cache policy python-catkin-pkg`)

Samu24, 2021-01-11:
My `python-catkin-pkg` version is `0.4.23-100`

Samu24, 2021-01-13:
I managed to solve the issue:
I was using a workspace management tool that required installation of catkin_tools from source. It looks like this installation was conflicting with the new release as the workspace does compile without issues after I removed the tool.

closed 2021-01-13 by Samu24