osrf/subt #731: No RGB data for new Dome/release 2020-12-16

m3d, 2020-12-17:
I am testing new release 2020-12-16 (locally), primarily to check Kloubak K2 lights, but I am not getting any RGB data there. Now the same for SSCI X4 config 2 - was there some rename of topics or some other change we should be aware of? I will check it also on CloudSim to have another reference.
thanks
   Martin

m3d, 2020-12-17:
p.s. just confirmed (robotika, `ver95artfdomecs1`, `24e784c3-4e2b-4a60-a25b-636f541a9dae`) - the drone passed backpack and did not report anything (it will be 100% when the drone returns and I will check the log files). So is it expected that the old docker image should work with the latest release?

nkoenig, 2020-12-17:
I'm seeing RGB data with the Kloubak. Here is what I'm running:

```
ign launch competition.ign worldName:=simple_cave_01 circuit:=cave robotName1:=R1 robotConfig1:=ROBOTIKA_KLOUBAK_SENSOR_CONFIG_2
```

```
rosrun rviz rviz
```

Then I can visualize the point cloud and RGBD image.

Do you see any error messages on the console?

m3d, 2020-12-17:
`36             rosmsg.image       0 |     0 |   0.0Hz`
i.e. no image data and the drone returned without scoring
[image]

m3d, 2020-12-17:
@nkoenig thanks - note that we are using docker images for the simulation and the bridge

m3d, 2020-12-17:
when I run `rostopic list | grep image` in the robot docker image I do not see any `compressed` topic, is it expected?

nkoenig, 2020-12-17:
The problem is the bridge docker image compiled `ros-ign` from source instead of using the debians. This cause image compression to be left out.

Fix is here: https://github.com/osrf/subt/pull/732

New docker image have been built and uploaded to cloudsim and dockerhub.

m3d, 2020-12-17:
Great! Thanks for quick response and fix! :)
```
root@md-ThinkPad-P50:/osgar-ws# rostopic list | grep image
/A150L/front/image_raw
/A150L/front/image_raw/compressed
/A150L/front/image_raw/compressed/parameter_descriptions
/A150L/front/image_raw/compressed/parameter_updates
/A150L/front/image_raw/compressedDepth
/A150L/front/image_raw/compressedDepth/parameter_descriptions
/A150L/front/image_raw/compressedDepth/parameter_updates
/A150L/front/image_raw/theora
/A150L/front/image_raw/theora/parameter_descriptions
/A150L/front/image_raw/theora/parameter_updates
/A150L/front/optical/image_raw
```
i.e. locally I can see the `compressed` image topics and I will check it also on CloudSim.

m3d, 2020-12-17:
now it works also on CloudSim:
[image]
thanks a lot
   m.

closed 2020-12-17 by m3d