osrf/subt #737: Could not find a package configuration file provided by "ignition-launch3"

pauljurczak, 2020-12-31:
`ignition-launch3` seems to be a new package not covered by installation instructions at https://github.com/osrf/subt/wiki/Catkin%20System%20Setup. I added it manually with `sudo apt install libignition-launch3-dev` after getting this error:
```
CMake Error at subt/subt_ign/CMakeLists.txt:37 (find_package):
  By not providing "Findignition-launch3.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "ignition-launch3", but CMake did not find one.

  Could not find a package configuration file provided by "ignition-launch3"
  with any of the following names:

    ignition-launch3Config.cmake
    ignition-launch3-config.cmake

  Add the installation prefix of "ignition-launch3" to CMAKE_PREFIX_PATH or
  set "ignition-launch3_DIR" to a directory containing one of the above
  files.  If "ignition-launch3" provides a separate development package or
  SDK, be sure it has been installed.
```

Perhaps it could be added to the sequence in *1, Setup and install dependencies*.

peci1, 2021-01-04:
That's weird:

```
$ apt show ignition-dome | grep launch

Depends: libignition-cmake2-dev, libignition-common3-dev, libignition-fuel-tools5-dev, libignition-gazebo3-dev, libignition-gui4-dev, libignition-launch3-dev, ...
```

So libignition-launch3-dev should have been installed automatically...

pauljurczak, 2021-01-04:
Thanks for checking it out. It was my error - I skipped a line in installation procedure.

closed 2021-01-04 by pauljurczak