osrf/subt #739: Docker: tutorial for adding CUDA?

peci1, 2021-01-06:
It is apparent many teams are using CUDA in their solutions, but the official Dockers do not contain the CUDA libraries. I understand that - everyone needs different versions etc.

However, as adding CUDA support is not really straightforward, I'd suggest to add a tutorial for adding it. Before we figured out how to "append" it to the subt-virtual-testbed images, we basically copy-pasted the subt dockerfiles into our dockerfiles based on nvidia/cuda, which is not really the best approach. Now that I have a reliable way of adding CUDA, we can base our images off of the official subt images. I guess other teams might benefit from that, too (and I guess many teams have already solved this issue internally, but I haven't seen it publicized anywhere).

This is my approach with which I tested Tensorflow 1.15 and Pytorch using CUDA: https://gist.github.com/peci1/0e8248a9e3045348549708e82b17e975 . I based the image off of the cloudsim_bridge_latest image, because subt-virtual-testbed:latest looks a bit weird at the moment, with outdated ros_ign (I'll consider changing it in my gist once that's fixed).

I haven't found any existing tutorial where this kind of information would actually fit, though... Maybe somewhere in subt_hello_world? But that would be pretty much buried in those lengthy posts...

If anyone's interested, the key thing that needs to be done (except installing the CUDA toolkit) is:

```Dockerfile
ENV NVIDIA_DRIVER_CAPABILITIES=all
```

That's needed because SubT images are based on image `opengl:1.0-glvnd-devel-ubuntu18.04`, which sets `NVIDIA_DRIVER_CAPABILITIES=graphics,compat32,utility`. CUDA requires `compute`, but that's missing.

adlarkin, 2021-01-14:
CUDA is used for [osrf/subt_hello_world](https://github.com/osrf/subt_hello_world) as well. One possible solution is to switch the base image in Dockerfiles from `nvidia/opengl:1.0-glvnd-devel-ubuntu18.04` to `nvidia/cudagl:X.X-devel-ubuntu18.04`, where `X.X` is the CUDA version the user needs (the CUDA version would depend on the user's GPU). So, for example, we could do something like replace the [base image in subt_shell](https://github.com/osrf/subt/blob/master/docker/subt_shell/Dockerfile#L1-L2) with the [base image in hello_world_common](https://github.com/osrf/subt_hello_world/blob/master/docker/hello_world_common/Dockerfile#L9-L12).

Would this be a sufficient solution?

peci1, 2021-01-14:
Oh, don't do that, please :) I think it's great everybody can decide on the version of CUDA he/she needs. Some tools like tensorflow 1 from pip need an exact minor number of CUDA version and won't work with any newer or older ones (just patch versions are okay to change).

adlarkin, 2021-01-14:
Well, if we do what I suggested in my previous message, the version can be specified through a Docker build arg. But if you don't want a change like that, then I'm not sure what you're looking for - are you just wondering if there's any documentation somewhere that shows how to integrate CUDA with the official subt docker images?

peci1, 2021-01-14:
> the version can be specified through a Docker build arg

You mean using an ARG in the FROM clause? It works, but looks kind of non-standard to me.

What I'm asking for is rather placing the CUDA-enabling piece of code somewhere where people will find it in case they want to base their image on some non-CUDA base image (be it cloudsim images or whatever else). Even as a commented-out section in places like:

https://github.com/osrf/subt/blob/51cb515b51ff4660b3d3d1a1eaf4042624a2c847/docker/cloudsim_bridge/Dockerfile#L127-L129

adlarkin, 2021-01-14:
:+1: yeah I don't think we have CUDA instructions anywhere, like you mentioned... I guess this is just something that we overlooked when initially setting up the Docker images :upside_down_face: I'm not sure where the best place to put something like this would be. Part of me is inclined to leave this issue open for now, because it allows others to see this conversation which includes a few proposals/workarounds... but eventually, it would be nice to mention this somewhere in the repository and/or wiki.

I'm not sure how most teams are incorporating CUDA, but when we were working on `subt_hello_world`, we used the `osrf/subt-virtual-testbed` as a way get access to the models, and then had another Dockerfile that was based on CUDA which had solution-specific code (a better description of this workflow can be found in the README [here](https://github.com/osrf/subt_hello_world/tree/master/docker)). Now, as you already mentioned originally in this issue, this requires a lot of copy/pasting and probably is not the optimal way of going about this. I like the approach you mentioned of using a single Dockerfile with the base image being `osrf/subt-virtual-testbed`, and then installing CUDA afterwards. I believe these are the "official" CUDA install instructions: https://docs.nvidia.com/cuda/cuda-installation-guide-linux/index.html. Perhaps we can mention in the wiki somewhere that you can do something like have a Dockerfile with a base image of `osrf/subt-virtual-testbed`, and then go through those instructions to install CUDA. Or, if that's a problem for users, they can try incorporating something like [nvidia/cudagl](https://hub.docker.com/r/nvidia/cudagl).

peci1, 2021-01-14:
Hmm, going through the subt_hello_world/docker directory, I now think it could be posible to change the hello_world_common image to be based on e.g. cloudsim_bridge, add CUDA according to the instructions I posted, and then add the rest of the subt_hello-specific code. This way, you'd get rid of the copy-pasting, you'd have a nice place to have these CUDA install instructions, and the subt repo would not need to be compiled as a part of the solution container (because it would be already compiled in the base image). And this common Dockerfile could be also filled with more tips for the teams, e.g. mentioning the cudagl image in case they need some different kind of setup etc. In the end, if I remember correctly, when creating our development pipeline, I used this common dockerfile as an inspiration.