osrf/subt #740: Dockerhub: update overview page and document sources for the docker images

peci1, 2021-01-06:
The overview page on dockerhub is no longer helpful: https://hub.docker.com/r/osrf/subt-virtual-testbed . It contains outdated links to bitbucket. The code snippet there seems also pretty outdated.

I was also digging around the docker images, and figured out that I can't tell which image is generated by which Dockerfile. I managed to come out with the obvious ones:

- cloudsim_bridge_latest -> github.com/osrf/subt/docker/cloudsim_bridge/Dockerfile
- cloudsim_sim_latest -> github.com/osrf/subt/docker/cloudsim_sim/Dockerfile
- models_latest -> github.com/osrf/subt/docker/subt_models/Dockerfile

But what are **subt_solution_latest** and **latest** generated from? subt_solution_latest seems to be especially weird - updated on Dec 17, but contains stuff like `/bin/sh -c apt-get update -qq &&  apt-get install -y -qq     ignition-blueprint     ros-melodic-ros1-ign  && apt-get clean -qq`. @nkoenig ?

I think it would really be helpful to have such a mapping directly on the overview page, so that people do not have to dig around to find out which image is for what (and what exactly does it contain).

peci1, 2021-01-22:
Okay, with the latest dockerhub release, it seems evident that the `:latest` tag on Dockerhub is generated from github.com/osrf/subt/docker/subt_sim_entry/Dockerfile .

nkoenig, 2021-05-05:
The dockerhub page has been updated: https://hub.docker.com/repository/docker/osrf/subt-virtual-testbed

closed 2021-05-05 by peci1