osrf/subt #741: Dockerhub: weird contents of images

peci1, 2021-01-06:
I was examining the contents of the images on dockerhub (as of Jan 6 2020).

I found these images:

- cloudsim_bridge_latest 7a840688d96e
- subt_solution_latest f8ba99591b71
- cloudsim_sim_latest c82e287eb5fc
- latest 83ff08ec4317
- models_latest 2f6c0bfa02ce

But [release notes page](https://github.com/osrf/subt/wiki/release_notes) lists:

- latest 83ff08ec4317
- cloudsim_sim_latest c82e287eb5fc
- cloudsim_bridge_latest 51652160ed4aa7
- models_latest 2f6c0bfa02ce

You can see that cloudsim_bridge_latest has a different digest on dockerhub and in release notes. If you try to click the [cloudsim_bridge_latest link](https://hub.docker.com/layers/osrf/subt-virtual-testbed/cloudsim_bridge_latest/images/sha256-51652160ed4aa74cb9ec97f16aa00aaa301333cfc74097d084a648c321b4c1d3?context=repo), you get a not-found page.

So this is **weirdness nr. 1**.

**Weirdness nr. 2**:

    $ docker run --pull --rm -it --entrypoint '' osrf/subt-virtual-testbed:latest /bin/bash -c 'ls /build*'
    /build_citadel.sh  /build_ign.sh

```
$ docker run --pull --rm -it --entrypoint '' osrf/subt-virtual-testbed:latest /bin/bash -c 'cat /home/developer/subt_ws/src/ros_ign/ros_ign_bridge/CMakeLists.txt | grep -C3 find_package.ignition-transport'
```

```CMake
# Default to Citadel, support Blueprint
if ("$ENV{IGNITION_VERSION}" STREQUAL "blueprint")
  find_package(ignition-transport7 REQUIRED)
  set(IGN_TRANSPORT_VER ${ignition-transport7_VERSION_MAJOR})

  find_package(ignition-msgs4 REQUIRED)
--

  message(STATUS "Compiling against Ignition Blueprint")
else()
  find_package(ignition-transport8 REQUIRED)
  set(IGN_TRANSPORT_VER ${ignition-transport8_VERSION_MAJOR})

  find_package(ignition-msgs5 REQUIRED)
```

```
$ docker run --pull --rm -it --entrypoint '' osrf/subt-virtual-testbed:latest /bin/bash -c 'apt-cache policy ros-melodic-ros-ign'
ros-melodic-ros-ign:
  Installed: (none)
  Candidate: 0.9.3-3bionic
```

So, osrf/subt-virtual-testbed:latest has outdated ros_ign. And looking at the "Dockerfile preivew" that dockerhub shows in image details, this image isn't based on models_latest.

**Weirdness nr. 3**: subt_solution_latest - what is it???

```
$ docker run --pull --rm -it --entrypoint '' osrf/subt-virtual-testbed:subt_solution_latest /bin/bash -c 'apt-cache policy ignition-dome; apt-cache policy ignition-blueprint'
N: Unable to locate package ignition-dome
ignition-blueprint:
  Installed: 1.0.0-1~bionic
  Candidate: 1.0.0-1~bionic
```

**Weirdness nr. 4**: cloudsim_sim_latest and cloudsim_bridge_latest contain gazebo3:

```
$ docker run --pull --rm -it --entrypoint '' osrf/subt-virtual-testbed:cloudsim_sim_latest /bin/bash -c 'apt-cache policy libignition-gazebo3'
libignition-gazebo3:
  Installed: 3.5.0-1~bionic
  Candidate: 3.5.0-1~bionic
```

This has already been reported and a plain rebuild of the images would fix that: https://community.gazebosim.org/t/apt-install-ignition-dome-installs-ign-gazebo3/772/4 .

**Minor weirdness nr. 5**: models_latest shows this in the "dockerfile preview viewer" on dockerhub:

    /bin/sh -c apt-get update &&  apt-get install -y     ignition-citadel  && apt-get clean

As it is removed in the last step, it's okay, but the dockerfile on github already uses dome for this one-off task... And this has also affected cloudsim_sim_latest and cloudsim_bridge_latest which are based on this outdated version of models_latest. It's weird just because this change on github was done just a few hours after the image was built...

peci1, 2021-01-12:
Also, osrf/subt-virtual-testbed:cloudsim_sim_latest doesn't have ros-melodic-ros-ign installed (although the Docker file has it).

peci1, 2021-01-22:
Good, the latest update on dockerhub cleared this a lot. models_latest, cloudsim_sim_latest and cloudsim_bridge_latest now contain the expected content declared in their respective dockerfiles as of today.

The `latest` tag has a glitch - it has been built upon an older version of the models_latest base image (docker says these layers are 5 weeks old).

Thanks for the update.

m3d, 2021-01-22:
p.s. please also fix the date of release in the notes:
https://github.com/osrf/subt/wiki/release_notes
(now 2021-12-16)
thanks m.

nkoenig, 2021-02-01:
There was a release on Dec. 16, 2020 and another release on Jan 21, 2021.

peci1, 2021-02-01:
Thank you for the info. But it is still unclear to me how should we treat the digests of docker images linked from the release notes. If you click any of the the digests from the latest release, it doesn't work because the images were updated after the release notes were created. So there's now no way to get the exact digests linked from release notes, because they have already disappeared from dockerhub...

nkoenig, 2021-03-02:
I don't know of a way to improve historical records unless we uniquely tag each release. Is this something you would like to see?

peci1, 2021-03-02:
I think it's not needed to uniquely tag each release. I see two ways how to make things more consistent:

1. Edit the digests for the latest release on the Releases wiki page every time you upload new Docker images. This way, the digests from the latest release would always be valid.
2. Start making minor or patch releases, which would be official SubT releases with only bugfixes. And mention these minor releases on the Releases wiki page. This would again result in having the latest release with valid digests. And would not require "rewriting history" as the first way does.

peci1, 2021-03-03:
@nkoenig It seems the dockerfiles uploaded to dockerhub are still being built from different dockerfiles than those present in this repo. That is confusing. If it is expected that the official images are built from different dockerfiles, please state that officially somewhere (e.g. in the published Dockerfiles). Accidentally, I checked the images right about the time you uploaded the new ones, and what I write holds for both the previous and the new version.

Namely, the cloudsim_sim_latest image doesn't build upon the models_latest image. Instead, it contains a `COPY` instruction followed by `RUN sudo chown -R`. It's not clear where do these instruction come from, but it's clear they pull in the model database. More importantly, these instructions greatly increase the size of the resulting image:

[image]

Instead of pulling models_latest, they add two layers, each of 1.6 GB download size (and 4.3 GB unpacked size).

peci1, 2021-03-03:
Ahh, after reviewing the latest uploaded docker images from docker hub, it seems that cloudsim_sim_latest is actually based on models_latest, so the copy&chown is a part of the models_latest image...