osrf/subt #750: Cloudsim: Backpack in staging area

peci1, 2021-01-12:
In simulation e097b600-5ff5-4ce0-a687-8079f46d0815, there was a backpack in staging area. It was world Cave Circuit 07. Another run in the same world does not show the backpack. But in the mentioned run, it was really there, as it generated events between time 23 and 51, which would be impossible with the backpack in correct position.

[image]

The generated events:

```YAML
- event:
  id: 6
  type: detect
  time_sec: 23
  detector: backpack_1
  robot: X1
  state: enter
  extra:
    type: artifact_proximity
- event:
  id: 7
  type: detect
  time_sec: 23
  detector: backpack_1
  robot: X2
  state: enter
  extra:
    type: artifact_proximity
- event:
  id: 8
  type: detect
  time_sec: 23
  detector: backpack_1
  robot: X3
  state: enter
  extra:
    type: artifact_proximity
- event:
  id: 9
  type: detect
  time_sec: 23
  detector: backpack_1
  robot: uav2
  state: enter
  extra:
    type: artifact_proximity
- event:
  id: 10
  type: detect
  time_sec: 23
  detector: backpack_1
  robot: uav4
  state: enter
  extra:
    type: artifact_proximity
- event:
  id: 11
  type: detect
  time_sec: 23
  detector: backpack_1
  robot: uav5
  state: enter
  extra:
    type: artifact_proximity
- event:
  id: 12
  type: detect
  time_sec: 44
  detector: backpack_1
  robot: X3
  state: exit
  extra:
    type: artifact_proximity
- event:
  id: 13
  type: detect
  time_sec: 46
  detector: backpack_1
  robot: X2
  state: exit
  extra:
    type: artifact_proximity
- event:
  id: 14
  type: detect
  time_sec: 51
  detector: backpack_1
  robot: X1
  state: exit
  extra:
    type: artifact_proximity
```

peci1, 2021-01-13:
Now again: 
[image] 

If you'll hurry up, you'll catch the sim still live, it has started about an hour ago. ID 81ce8cb5-a9ad-4b67-ae79-f86b5126c57c

nkoenig, 2021-03-02:
Have you seen this again? I'm unable to reproduce it using a source build or using docker images.

peci1, 2021-03-02:
No, Jan 13 was the last time we saw this. Simulations run on Jan 22 did not contain the backpack and neither do any later simulations. I'm closing this for now, and will reopen if it happens again.

closed 2021-03-02 by peci1