osrf/subt #770: Catkin build problem

pauljurczak, 2021-01-29:
I'm building Catkin version of the newest release v2021-01-21. I have two PCs with Ubuntu 18.04.5. It builds fine on one, but produces linker error on another. There are differences in development libraries between these systems, but I can't find out potential version mismatch. here is the error:
```
[ 89%] Linking CXX executable /home/paul/subt_ws/devel/lib/subt_ign/validate_visibility_table
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFTileRowSize@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFReadDirectory@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFFieldTag@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFCurrentDirectory@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFLastDirectory@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFIsTiled@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFIsByteSwapped@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFDefaultStripSize@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFComputeStrip@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFReadTile@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFWriteScanline@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFGetField@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFScanlineSize@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFFieldWithTag@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFTileSize@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFWriteDirectory@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFFieldName@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFFieldReadCount@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFGetTagListEntry@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFClose@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFClientOpen@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFSetTagExtender@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFGetTagListCount@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFGetFieldDefaulted@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFSwabArrayOfLong@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFFieldPassCount@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFSetSubDirectory@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFStripSize@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFFieldDataType@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `_TIFFDataSize@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFReadRGBAImage@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFReadEncodedStrip@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFSetField@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFMergeFieldInfo@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFDataWidth@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFSetDirectory@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libfreeimage.so.3: undefined reference to `TIFFReadEXIFDirectory@LIBTIFF_4.0'
collect2: error: ld returned 1 exit status
subt/subt_ign/CMakeFiles/validate_visibility_table.dir/build.make:189: recipe for target '/home/paul/subt_ws/devel/lib/subt_ign/validate_visibility_table' failed
make[2]: *** [/home/paul/subt_ws/devel/lib/subt_ign/validate_visibility_table] Error 1
CMakeFiles/Makefile2:6912: recipe for target 'subt/subt_ign/CMakeFiles/validate_visibility_table.dir/all' failed
make[1]: *** [subt/subt_ign/CMakeFiles/validate_visibility_table.dir/all] Error 2
```
The libraries I suspected are identical between my two PCs:
```
(base) paul@desktop:~/subt_ws$ apt list --installed libfreeimage*
Listing... Done
libfreeimage-dev/bionic-updates,bionic-security,now 3.17.0+ds1-5+deb9u1build0.18.04.1 amd64 [installed]
libfreeimage3/bionic-updates,bionic-security,now 3.17.0+ds1-5+deb9u1build0.18.04.1 amd64 [installed,automatic]
(base) paul@desktop:~/subt_ws$ apt list --installed libtiff*
Listing... Done
libtiff-dev/bionic-updates,bionic-security,now 4.0.9-5ubuntu0.3 amd64 [installed,automatic]
libtiff5/bionic-updates,bionic-security,now 4.0.9-5ubuntu0.3 amd64 [installed,automatic]
libtiff5-dev/bionic-updates,bionic-security,now 4.0.9-5ubuntu0.3 amd64 [installed,automatic]
libtiffxx5/bionic-updates,bionic-security,now 4.0.9-5ubuntu0.3 amd64 [installed,automatic]
```
Are there any other libraries or settings I should check?

pauljurczak, 2021-01-31:
I can build the previous version downloaded on Dec. 31, 2020 with no problems on that PC. What changes affected `libfreeimage` or `libtiff` since then?

nkoenig, 2021-02-02:
This is a strange error. The `validate_visibility_table.cc` file moved to into an `app` subdirectory in `subt_ign/src`. The linking process did not change. Your PC with the error might have some lingering libraries. Can you try making sure that you have the latest `osrf/subt` code and maybe clean out your `build` and `devel` folder in your catkin workspace?

pauljurczak, 2021-02-10:
I completely deleted catkin workspace, pulled the new version and built. It didn't help. Do you have an estimate when the next version will be released? Maybe it will build automagically.

pauljurczak, 2021-02-27:
The problem was created by Miniconda, which had its own copy of `libtiff.so.5`. Uninstalling Miniconda solved the problem. The mechanism is not clear to me, because:
```
paul@desktop:~$ ldd /usr/lib/x86_64-linux-gnu/libfreeimage.so.3
....
	libtiff.so.5 => /usr/lib/x86_64-linux-gnu/libtiff.so.5 (0x00007f2646a4f000)
....
```
which indicates that `/usr/lib/x86_64-linux-gnu/libtiff.so.5` should have been used by catkin build. Apparently another version was used instead.

closed 2021-02-27 by pauljurczak