osrf/subt #782: Tunnel Tile 6 missing in "ign fuel list"

peci1, 2021-02-05:
```
$ ign fuel list -v4 -o openrobotics -t model | grep -i "tunnel tile" | sort
    ├── constrained tunnel tile short
    ├── constrained tunnel tile short
    ├── constrained tunnel tile tall
    ├── rough tunnel tile 4-way intersection
    ├── rough tunnel tile 4-way intersection
    ├── rough tunnel tile 90-degree turn
    ├── rough tunnel tile 90-degree turn
    ├── rough tunnel tile ramp
    ├── rough tunnel tile ramp
    ├── rough tunnel tile straight
    ├── rough tunnel tile straight
    ├── rough tunnel tile vertical shaft
    ├── tunnel tile 1
    ├── tunnel tile 1
    ├── tunnel tile 2
    ├── tunnel tile 2
    ├── tunnel tile 3
    ├── tunnel tile 4
    ├── tunnel tile 5
    ├── tunnel tile 5
    ├── tunnel tile 7
    ├── tunnel tile blocker
    ├── tunnel tile blocker
```

However, "Tunnel Tile 6" is used e.g. in tunnel_circuit_practice_03.sdf world.

Browsing the online database at subtchallenge.world, I can find this model. But `ign fuel list` doesn't know about it for some reason.

nkoenig, 2021-03-02:
I think this has been fixed. I see Tunnel Tile 6 in the results.

peci1, 2021-03-02:
I've reported bug https://github.com/ignitionrobotics/ign-fuel-tools/issues/168 . It seems that `ign fuel list -v4 -o openrobotics -t model` only searches the cached models. This is what I see on a relatively new machine that hasn't yet run the download all script:

```
$ ign fuel list -v4 -o openrobotics -t model
Fetching model list from https://fuel.ignitionrobotics.org...
Received model list (took 0ms).
https://fuel.ignitionrobotics.org
└── openrobotics
    ├── cave cavern split 02 type b
    ├── rescue randy
    ├── cave straight 04 type b
    ├── ctu_cras_norlab_absolem_sensor_config_2
    ├── rescue randy sitting
    ├── cave elevation type b
    ├── cave straight 01 type b
    ├── fiducial
    ├── x1 config 1
    ├── cave vertical shaft type b
    ├── samsung j8 black
    ├── ctu_cras_norlab_absolem_sensor_config_1
    ├── cave straight 05 type b
    ├── explorer_x1_sensor_config_1
    ├── cave 3 way 01 type b
    ├── cave starting area type b
    ├── jansport backpack red
    ├── breadcrumb node
    ├── cave straight 02 type b
    ├── artifact proximity detector
    ├── cave corner 01 type b
    ├── cave cap type b
    ├── cave straight type a
    ├── base station
    ├── cave corner 02 type b
    ├── explorer_x1_sensor_config_2
    ├── x1 config 2
    ├── cave cavern split 01 type b
    ├── construction cone
    └── cave straight 03 type b
```

I'm closing this issue because it is probably a problem in ign fuel (or its documentation).

closed 2021-03-02 by peci1