osrf/subt #783: Errors in Virtual STIX fuel models

peci1, 2021-02-05:
All Virtual STIX fuel models have invalid pose: 

    <pose>0 0 0 1.5707 0 0 0</pose>

that's 7 elements, but pose should always have 6 according to SDF spec.

Additionaly, the numbered Virtual STIX models have invalid XML in model.config: tag `<model>` is closed twice (once on the depend line and once at the end of the document):

```xml
<?xml version="1.0"?>

<model>
  <name>Edgar Mine Virtual STIX 1</name>
  <version>1.0</version>
  <sdf version="1.6">model.sdf</sdf>

  <author>
    <name>Cole Biesemeyer</name>
    <email>cole@openrobotics.org</email>
  </author>

  <description>
    A section of the Edgar Mine for use during the SubT Challenge Virtual STIX competition.
  </description>

  <depend><model><uri>https://fuel.ignitionrobotics.org/1.0/openrobotics/models/edgar mine virtual stix</uri></model></depend></model>

</model>

```

nkoenig, 2021-02-16:
The models have been fixed, and a [new release is out](e027b2260c476b907fd403f6682436313394a41bd6001a6f86691c5102698eea).

peci1, 2021-02-16:
Thanks for the fix, it looks good - with one remnant:

```
$ grep -r "0 0 0 1.5707 0 0 0" /home/developer/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/                                 
/home/developer/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/edgar mine virtual stix/11/model.sdf:            <pose>0 0 0 1.5707 0 0 0</pose>
```

nkoenig, 2021-03-02:
This one has been fixed as well.

closed 2021-03-02 by nkoenig