osrf/subt #789: Trouble with generating .dat file for custom world

mohradza, 2021-02-11:
I am trying to generate some simple custom worlds, but I am having trouble generating the <world>.dat file. I am following the tutorial from https://github.com/osrf/subt/wiki/World-Creation-Tutorial. 

First, I tried to create a simple world consisting of just the staging area, base_station, and artifact_origin. I was able to generate both the .sdf and .dot files using the tools listed in the tutorial. When I tried to generate the .dat file using the `ign launch visibility.ign worldName:=my_new_world` command, it just hung and never completed. 

Then I tried to start with an existing world model, so I loaded up simple_cave_01.sdf into ignition gazebo and added some sphere shapes to the staging area. Then I saved this file as a new .sdf. I followed the instructions to generate the .dot file with no issues, but when I tried to generate the .dat file it hung again. I let it run for several hours, and it did not complete. 

Has anyone had any success generating custom worlds or have any ideas on how I might debug this?

Thanks!

mohradza, 2021-02-12:
I found that you don't need to regenerate the the .dat file in order for added shapes / models to appear in the ignition sim. This is good enough for my application, as I am just trying to add obstacles for a vehicle to avoid.

I would still be interested in trying to figure out how to debug the issue mentioned above in the event that we do want to generate full custom worlds.

nkoenig, 2021-02-16:
Can you post the world file that you were trying to generate a `.dat` for?

nkoenig, 2021-03-02:
Closing due to lack of response.

closed 2021-03-02 by nkoenig