osrf/subt #800: Catkin System setup wiki: Needs a few updates

peci1, 2021-02-24:
I was just installing a fresh Ubuntu 18.04 system and noticed a few glitches in the https://github.com/osrf/subt/wiki/Catkin%20System%20Setup wiki.

- step 1 should also install `python-rosdep`
- somewhere between 2-4 should also be `sudo rosdep init && rosdep update`
- step 5 should also call `rosdep install --from-paths src --ignore-src -r` before building the WS
- the build command should be `catkin_make -DCMAKE_BUILD_TYPE=Release install` to be consistent with the docker images

Lower on the page, there is the ign fuel download command, which uses 16 threads. I think that's too much (you probably know better what is the best number, but it seemed to me that all `-j16` were already lowered to `-j8` in the dockers).

pauljurczak, 2021-02-26:
Additionally, step 1 should install `git`, which doesn't come with a standard Ubuntu distribution.

nkoenig, 2021-03-02:
Thanks, all changes have been made.

closed 2021-03-02 by peci1