osrf/subt #807: Missing pose_from_artifact_origin service in final qualification

m3d, 2021-03-03:
Hello,
it looks like the robots are not starting in just released final qualification world and in the rosout.log I can see
```
43.716000000 ERROR /A600LXM_proxy [/osgar-ws/src/osgar/subt/ros/proxy/ros_proxy_node.cc:395(Update)] [topics: /rosout, /A600LXM/cmd_vel] Failed to call pose_from_artifact_origin service, robot may not exist, be outside staging area, or the service is not available.
```
can you confirm that it is working for you?
thanks
   Martin/Robotika Team
p.s. simulation ver98octofinal (`23a54806-7204-4951-ba18-003a1f243efe`)

angelacmaio, 2021-03-03:
The fix is in PR #806, which was recently merged and should be deployed to Cloudsim soon.

m3d, 2021-03-04:
Thanks for the fix and new docker images - I can confirm that now both robots started on CloudSim. :)

closed 2021-03-04 by m3d